realsenseai/librealsense #6613: intel real sense sdk version 2.0软件打不开

woshixiaodoudouao, 2020-06-17:
程序没有一点反应，同样的安装包安装到别的电脑上，可以打开软件但是识别不了摄像头。

MartyG-RealSense, 2020-06-17:
Translation:  "There is no response from the program (Intel RealSense software 2.0). The same installation package is installed on other computers.  The software can be opened but the camera cannot be recognized".

Hello @woshixiaodoudouao   Are you using Windows 10 please?

您好@woshixiaodoudouao您正在使用Windows 10嗎？

woshixiaodoudouao, 2020-06-17:
> 转换：“程序没有响应（Intel RealSense软件2.0）。在其他计算机上安装了相同的安装包。可以打开该软件，但无法识别相机”。
> 
> 您好@woshixiaodoudouao 您正在使用Windows 10吗？
> 
> 您好@woshixiaodoudouao您正在使用Windows 10吗？

是的，都是win10，四台电脑只有一台能打开，很崩溃

MartyG-RealSense, 2020-06-17:
Original link: https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

請檢查Windows 10隱私設置，將對照相機硬件的訪問和對照相機應用程序的訪問均設置為“開”。

為此，如果在Windows屏幕底部的文本框中鍵入“ privacy”（在關閉PC的按鈕旁邊），則會找到“ Privacy Settings”選項。 從左側的選項面板中選擇“相機”類別，並確保“允許訪問此設備上的相機”和“允許應用程序訪問您的相機”都設置為“開”。

[image]

woshixiaodoudouao, 2020-06-18:
> Original link: [#3381 (comment)](https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789)
> 
> 請檢查Windows 10隱私設置，將對照相機硬件的訪問和對照相機應用程序的訪問均設置為“開”。
> 
> 為此，如果在Windows屏幕底部的文本框中鍵入“ privacy”（在關閉PC的按鈕旁邊），則會找到“ Privacy Settings”選項。 從左側的選項面板中選擇“相機”類別，並確保“允許訪問此設備上的相機”和“允許應用程序訪問您的相機”都設置為“開”。
> 
> [image]

谢谢啊，还有一个问题，有一个摄像头在工作几分钟以后，就会断开连接，把线拔了重新插就又好了，然后几分钟又断开连接了。

MartyG-RealSense, 2020-06-18:
If you can use the RealSenseViewer program, please look at the ASIC temperature in the Controls section of the options.  What is the temperature value when the camera disconnects please?

如果可以使用RealSenseViewer程序，請在選項的“控件”部分中查看ASIC溫度。 請在斷開相機連接時的溫度值是多少？

MartyG-RealSense, 2020-07-05:
Hello @woshixiaodoudouao   Do you still require assistance please?

您好@woshixiaodoudouao您仍然需要幫助嗎？

closed 2020-07-18 by MartyG-RealSense