realsenseai/librealsense #6616: any Guide doc?

ipinyulun, 2020-06-17:
hi, i been looking into the github readme here and even download it to read.
all the doc available here are aabout installing the module only.
but i cant find any guide of the realsense where we could achieve mapping and navigation as per this video below
[https://www.youtube.com/watch?v=LBMsWJJxLXQ&feature=youtu.be](https://www.youtube.com/watch?v=LBMsWJJxLXQ&feature=youtu.be)

is there actually any guide exist? or do i have to buy the hardware first?

MartyG-RealSense, 2020-06-17:
Hi @ipinyulun  You do not have to buy a RealSense camera first to ask questions.   Do you know which camera model you may be interested in, please?  This will help us to direct you to the right guides.  Or would you like links to a range of different guides so that you can evaluate which camera would be best for your work?

ipinyulun, 2020-06-18:
i already purchased both rs 435 and t265
and what i wanted is to have it work like the video. mapping and navigation.
may you please kindly show me where is the guide?

MartyG-RealSense, 2020-06-18:
Intel have published guides for using D435 and T265 together:

https://www.intelrealsense.com/depth-and-tracking-combined-get-started/

https://dev.intelrealsense.com/docs/depth-and-tracking-cameras-alignment?_ga=2.169671970.2094136317.1577013487-1521828152.1577013487

If you have further questions about using D435 and T265 together, please post them as a new issue on the GitHub instead of commenting here.  Thanks!

closed 2020-07-05 by MartyG-RealSense