realsenseai/librealsense #6618: can you offer other lib source url？

comwise, 2020-06-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 16。04  | 
| Kernel Version (Linux Only)    |  4.15.xx                                         | 
| Platform                                 | PC  |
| SDK Version                            |                            | 
| Language                            |  {C++/python }                          | 
| Segment			|  Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
the follow url download from china which is so low， can you offer other url？
such as github or all builded lib for kinetic or melodic
deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main
deb-src http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main

MartyG-RealSense, 2020-06-21:
I apologise for the delay in answering your question.

I believe you are asking for a different way to install Librealsense than using the Debian distribution packages.  Is that correct please?

If it is correct, then there are other ways to install Librealsense.

1.  If you want to install Librealsense and the RealSense ROS wrapper at the same time, you could try **ROS Installation Method 1** in the link below.  This installs realsense2_camera and Librealsense.

The version of Librealsense installed using this method may be older than the newest version on the SDK 'Releases' page though.

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

2.  Another way to install Librealsense easily over an internet connection is the **backend** method.  This method uses Librealsense source code.  It does not depend on Linux versions or kernel versions to work, and it does not need to be patched.

If you wanted to install SDK **2.35.2**, first go to the bottom of the release details for 2.35.2 and find the **Assets** list of files.  Download the **source code** zip file and then extract its contents.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.35.2

[image]

Once the zip file has been extracted, you can follow the instructions in the link below to install Librealsense through your internet connection without patching:

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

RealSenseSupport, 2020-07-02:
@comwise Sorry, currently we don't have other link to download the binary. Could you please try another network setup to see if any network connection issue? Thanks!

closed 2020-07-09 by dorodnic