realsenseai/librealsense #6619: Pyrealsense2 RuntimeError:

EvanZouQ, 2020-06-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | not sure (default) | 
| Operating System & Version |  Win 10 | 
| Platform                                 | PC |
| IDE                                          |Pycharm|
| SDK Version                            | 2.33. | 
| Language                                | python | 
| Segment			          | object detection   | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi everyone!
I have install pyrealsense2 using pip. But I try to use Pycharm to read the data captured by Intel D435i. I always get error as follows:

Traceback (most recent call last):
  File "D:/PycharmProject/IntelCamera/test.py", line 12, in <module>
    pipeline.start(config)
RuntimeError: 
Failed to resolve the request: 
	Format: Z16, width: 848, height: 480

The python codes is shown as follows;

import pyrealsense2 as rs
import numpy as np
import cv2

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
pipeline.start(config)

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        # Stack both images horizontally
        images = np.hstack((color_image, depth_colormap))

        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)

        key = cv2.waitKey(1)
        # Press esc or 'q' to close the image window
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break

finally:

    # Stop streaming
    pipeline.stop()

I am new to this field. So I don't know how to solve this problem. Thanks for your assistance.

MartyG-RealSense, 2020-06-17:
"Failed to resolve the request" may indicate that the SDK has been asked to perform an action that is does not understand.  An example of such a request may be defining a resolution or stream format that is not supported.  Your stream definition statements look correct though.

Could you try changing pipeline.start(config) to this please:

**profile = pipeline.start(config)**

EvanZouQ, 2020-06-17:
Dear Sir or Madam, I have try your code.  But it returns the same error. What should I try? Thanks a lot.

EvanZouQ, 2020-06-17:
> "Failed to resolve the request" may indicate that the SDK has been asked to perform an action that is does not understand. An example of such a request may be defining a resolution or stream format that is not supported or has a spelling mistake. Your stream definition statements look correct though.
> 
> Could you try changing pipeline.start(config) to this please:
> 
> **profile = pipeline.start(config)**

Dear Sir or Madam, I have try your code. But it returns the same error. What should I try? Thanks a lot.

MartyG-RealSense, 2020-06-17:
I researched the issue further.  There was a past case of a Python Pycharm user who experienced the same error with a similar script.  They were advised that the defined stream configuration may have been valid on a USB3 connection but not supported on USB2.

https://github.com/IntelRealSense/librealsense/issues/1656#issuecomment-389136061

Is your camera in a USB 3 port please, and if so, is it detected as USB 3?

EvanZouQ, 2020-06-17:
I have double checked. I can confirm that the USB is 3 and the Intel Viewer displays as USB 3.2. But it returns the result. Thanks.

MartyG-RealSense, 2020-06-17:
I wonder if you could try a simpler test script that uses config and see if the error still occurs.

import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
pipeline.start(config)

EvanZouQ, 2020-06-17:
Sorry for the delay.  I have tested your simple codes. It returns the same error"Failed to resolve the request"

And I have checked Intel Viewer.  The resolutions of stereo module and RGB camera are 848*480, 1280*720 respectively. I don't know do these resolution have certain effects?

MartyG-RealSense, 2020-06-17:
Higher resolutions may make streaming lag on some computers, but all supported resolutions should be recognised if the camera is on a USB 3 connection.

The absolute most basic state that the test script can be reduced to is a one-line script to test whether the pyrealsense2 module can be detected:

**import pyrealsense2 as rs**

If you still get an error from that then it may indicate a more fundamental prob;lem with the pyrealsense2 installation than just a stream configuration issue.

EvanZouQ, 2020-06-17:
There is no error when only run "import pyrealsense2 as rs". I think that I have successfully installed pyrealsense using pip. But when run the previous codes. It causes same error. Thanks.

MartyG-RealSense, 2020-06-17:
Let's test the camera with a couple more test scripts to confirm that pyrealsense2 is streaming properly.

First there is this simple example from the PyPi pip site:

***********

import pyrealsense2 as rs
pipe = rs.pipeline()
profile = pipe.start()
try:
  for i in range(0, 100):
    frames = pipe.wait_for_frames()
    for f in frames:
      print(f.profile)
finally:
    pipe.stop()

***********

And here is a more complex beginner script called **tutorial_1_depth** that renders the depth stream on-screen in the form of ASCII text-character art.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py

EvanZouQ, 2020-06-17:
Dear Sir or Madam, I have tested the tutorial_1_depth script. But it still returns the same error‘Failed to resolve the request: 
	Format: Z16, width: 848, height: 480

Into:
	Formats: 
	 Z16‘
Does this means that the pyrealsense2 is streaming inproperly? Some scripts have been tested and the results/error are the same. I have no idea about that. Thnaks.

MartyG-RealSense, 2020-06-17:
The same error has occurred for others, even when only streaming infrared:

https://github.com/IntelRealSense/librealsense/issues/3885

Have you tried updating the firmware driver of your D435i please?

EvanZouQ, 2020-06-17:
Dear MartyG, I hvae checked the specification which is shown as follows:
1) Name: Intel RealSense D435I, serial number: 919122070934, update serial number: 908323050036, firmware version: 05.12.05.00, USB type: 3.2
I think the firmware is up to date. Thanks.

MartyG-RealSense, 2020-06-17:
Thanks.  Yes, your firmware is up to date.

Did you find anything helpful in https://github.com/IntelRealSense/librealsense/issues/3885

EvanZouQ, 2020-06-17:
I think I have no such issues which are presented in the link. I have no progress on this error.  Thanks.

MartyG-RealSense, 2020-06-17:
Since the simple example scripts did not work for you, the problem is unlikely to be caused by your custom **config** stream configuration.

Researching further, I came across a past case I had handled where the 'could not resolve request' error disappeared after using the Dynamic Calibrator software of the SDK to recalibrate the camera. 
 The Dynamic Calibrator tool can be downloaded from here:

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

EvanZouQ, 2020-06-18:
Dear MartyG, I have tried Dynamic Calibrator tool. But it returens 'Failed to start calibration' and consequently the python script fails again.Thanks.

MartyG-RealSense, 2020-06-18:
"Failed to start calibration" can be a symptom of being on a USB 2 connection.  This is because the Dynamic Calibrator tool needs the camera to be on a USB 3 connection.  

A number of RealSense users have experienced problems with a failed calibration on the Dynamic Calibrator though even when plugged into USB 3.

https://github.com/IntelRealSense/librealsense/issues/3990

It also currently has problems with running on Ubuntu PCs that have the '5.3' kernel version.  You listed in your information at the start of the case that you were using Windows.  Can you confirm please that you ran the Dynamic Calibrator program on Windows?

EvanZouQ, 2020-06-18:
I can confirm that I ran Dynamic Calibrator on Window 10 and the USB is 3. But it fails. Thanks.

MartyG-RealSense, 2020-06-18:
Assuming that you are using Windows 10 and not Windows 7 (as your provided information at the top of the case says Win 10):

One thing we have not considered is if the problem might be with Pycharm, and if using a different Python compiler might make a difference.

EvanZouQ, 2020-06-19:
Dear MartyG， I have tried several compiler such as Pycharm, Jupiter and Visual Studio. But the same result "Failed to resolve the request" appear. 
So I consider whether the SDK version will affect this? My SDK version is 2.33.1. And could you please give me a guide on installing liberealsense and pyrealsense step by step?  I am wondering if I lost one step, which causes this error.

MartyG-RealSense, 2020-06-20:
The Librealsense version should not matter.  There does not need to be a close match between the SDK version and the Pyrealsense2 version.  The **Python** version that you are using (2.7 or one of the 3.x versions) has a strong influence on how the Pyrealsense2 wrapper should be installed though.

I checked this discussion and found that the Python version number that you are using has not been mentioned before.  Which Python version are you using, please?

EvanZouQ, 2020-06-20:
The Python version is 3.6.5（By Anaconda）. Thanks

MartyG-RealSense, 2020-06-20:
I have some notes from when I installed pyrealsense2 on Python 3.7 using the pip install method, so I will try to recreate what I did.  I based my method on the instructions in the link below.

https://www.liquidweb.com/kb/install-pip-windows/

1.  Download the file **get-pip.py**.   This is to install the **pip** file installation system that is necessary to use the **pip install pyrealsense2** method of setting up pyrealsense2.

Right-click on the link below and select 'Save link as' from the menu to download the file to your Windows computer:

https://bootstrap.pypa.io/get-pip.py

2.  Move the downloaded file to the root folder of your Python installation (e.g 'Python36' if you have Python 3.6 installed).

3.  Open a Windows **command prompt** window.  You can do this by typing 'command' into the text box at the bottom your Windows screen, next to the button in the corner where you shut down the PC.

4.  Change directory in the command prompt window to the location of the Python root folder where you placed the downloaded file, using the **cd** command. 

[image]

5.  Input this command into the command prompt window to run the **python get-pip.py** file:

**python get-pip.py**

Running this file should install the **pip** installation system (not pyrealsense2).  

6.  Use the **cd** command to change the command prompt directory to the folder where the command prompt says that Pyrealsense2 was installed to.  On Python 3.6 it will probably be **Python36 > Scripts**

7.  Input **pip** into the Command Prompt window to check whether pip has been installed successfully.

8.  If it has, then input **pip install pyrealsense2** to install Pyrealsense2.

EvanZouQ, 2020-06-20:
Thanks for your reply.  I will try it. Thanks.

EvanZouQ, 2020-06-20:
Dear MartyG, I have tired to update all the versions of python, opencv, opecv-contrib and pyrealsense.  But it still returns the same error. How can I fix this error? Sorry for bothering you during weekend. Thanks.

MartyG-RealSense, 2020-06-21:
Next, try inserting this line directly before your pipeline start instruction:

**profile = config.resolve(pipeline)**

This tries to 'resolve' the configuration to find a matching device and stream profiles, testing for conflicts.

So the script would look like this:

profile = config.resolve(pipeline)
pipeline.start(config)

More information about testing the config can be found here:

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.config.html

EvanZouQ, 2020-06-21:
Thanks for your reply. However, it deosn't work. It returns the same result 'Failed to resolve the request'. I have no idea on this error. Thanks again.

MartyG-RealSense, 2020-06-21:
Let's eliminate another possibility by checking if Advanced Mode is enabled on your camera.  Please go into the RealSense Viewer with your camera plugged in and click on the 'More' option at the top of the options side-panel.  Then click on the menu item called '**Advanced Mode**'.  If Advanced Mode on your camera is turned off, a pop-up window will ask if you want to turn it on.  Say yes.

Then run your Pyrealsense2 script and see if anything has changed.

Once Advanced Mode is set to On, the camera should stay like that even when the Viewer has been closed.

EvanZouQ, 2020-06-21:
It still returns the same result. Thanks.

MartyG-RealSense, 2020-06-21:
Have you got the files **realsense2.dll** and **pyrealsense2.pyd** in the same folder as your **test.py** script, please?

EvanZouQ, 2020-06-21:
Sorry I don't hear of the two files. And where can I check the location of the two files. Thanks.

MartyG-RealSense, 2020-06-21:
They can both be found on Windows inside the Librealsense folder location below:

**C: > Program Files (x86) > Intel RealSense SDK 2.0 > bin > x64**

[image]

If you do not have this folder, it can be installed on Windows by installing the pre-built binary version of Librealsense from the 'Releases' web-page.

[image]

EvanZouQ, 2020-06-21:
Dear MartyG, Thanks for your reply. When I copy the two files into the same folder of my test.py, it causes another error  ''import pyrealsense2 as rs
ImportError: DLL load failed: The specified module could not be found.''  I have tried several ways which is provided by Google. However, I fail to solve this problem. Thanks again.

MartyG-RealSense, 2020-06-21:
Checking the Windows installation instructions, I see what my mistake may be.   If the two DLL files are obtained from the folders created with the Windows pre-built installer from the Releases page then they create DLLs that are compatible with Python 2.7, and will cause the DLL load failed error if used with Python 3.  I do apologise.

MartyG-RealSense, 2020-06-21:
As you have already installed pyrealsense2 with **pip install pyrealsense2**, the pyrealsense2.pyd file that will work with Python 3 should already be on your computer now, though it is a little awkward to find.

1.  Go to your Python36 folder.

[image]

2.  In the directory window's search box, put in this exact search word:

**pyrealsense2.pyd**

[image]

Python 3 has an unusual naming system for the pyd file compared to Python 2.7, where it is just called pyrealsense2.pyd.  On Python 3 it will appear in the search with a longer name.

[image]

3.  Right-click on the file in the search results and select **Copy** from the menu to copy the pyd file.

[image]

4.  Go to the folder where your script is, right-click inside the folder and select **Paste** to paste a copy of the file into the folder.  Delete the **pyrealsense2.pyd** file that we were using earlier.

5.  See if your Pyrealsense2 script will now run.

MartyG-RealSense, 2020-06-21:
I am away for 7 hours from the time of writing this now as it is the end of my day.  I will pick up your next comment when I return.  Good luck!

3773nothview, 2020-06-26:
We are experiencing the exact same problem described in this thread, however the error occurs randomly, ~ 25% of the time.

3773nothview, 2020-06-26:
EvanZouQ , I am curious to know what calibration methods you attempted prior to encountering this message.   In our case, we have tried to use several of the calibration tools provided by Intel, on a Win 10 machine, prior to installing the camera on our Ubuntu platform.   Perhaps we can find something in common between our initial setup efforts that is causing the problem (eg: some residual in the calibration parameters / flags of the camera)?

EvanZouQ, 2020-06-26:
Dear 3773nothview, I have no idea if I lost something during installation. I followed the instruction. But I doesn't work. Therefore, the hareware of my PC maybe the main reason. So I gave up and try to do offline manipulation.Thanks

MartyG-RealSense, 2020-07-05:
Hi @EvanZouQ   Do you require further assistance with this case please, or is it okay to close it?  Thanks!

3773nothview, 2020-07-05:
It's remarkable to me, how many issues I've encountered trying to get realsense to work.  I can often find numerous threads discussing very similar issues, but the vast majority never seem to be resolved, because many people, myself included,  simply don't have days or even weeks  to burn trying to debug the platform.  So these tickets just get closed.   I think Intel could do more to proactively look at these collective setup / configuration issues and find the root cause(s).

MartyG-RealSense, 2020-07-05:
There are installation shortcuts for librealsense, pyrealsense and ROS.  It can admittedly take some research to find them on your own though.

For librealsense, use the **backend** installation method to install the SDK over an internet connection without dependency on Linux version, kernel version or patching.

For Python, including the **BUILD_PYTHON_BINDINGS** term in the librealsense CMake build statement so that pyrealsense2 is built along with the SDK.

For ROS, choosing the installation method that installs librealsense and ROS together (though the SDK version installed may be older than the most recent one).

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

dorodnic, 2020-07-09:
@3773nothview issues are not closed without investigation and response. I can ensure you this issue is not common. 
For this issue, @EvanZouQ - could you please add `rs.log_to_console(rs.log_severity.debug)` before creating the pipeline? Also, if its possible to attach output of `rs-enumerate-devices` from SDK package it can help us understand what is different about this platform.

MartyG-RealSense, 2020-07-18:
Case closed due to no further comments received.

closed 2020-07-18 by MartyG-RealSense

EvanZouQ, 2020-07-31:
I am sorry that I forgot to reply.  This can be closed. Thanks for your effort.