realsenseai/librealsense #6620: Librealsense Nanopi NEO4 install

noblegasses, 2020-06-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   | 2.33.1 | 
| Operating System & Version |   Linux Ubuntu 18.04 (FriendlyDesktop) | 
| Kernel Version (Linux Only)    |  4.4.179                                         | 
| Platform                                 | NanoPi NEO4   |
| SDK Version                            |  2.0                       | 
| Language                            |  python/C++                       | 
| Segment			|  others                           | 

### Issue Description
I am trying to get a 4.35i working on a nanopi NEO4 for a computer vision project. I have tried multiple installs using the pre-built packages and the manual install. Currently, on the pre-built install it fails to locate librealsense2-dkms, in a manual install the kernel patch: `./scripts/patch-realsense-ubuntu-lts.sh` fails because 4.4.179 is not reported. if these commands are ignored, and the installation continues, either Realsense-viewer cannot detect the camera, or it launches with a rockchip error, and fails to load any stream other than the depth stream and the IMU data. Any advice on how to get this working? I know the basics of linux but not much more, so

MartyG-RealSense, 2020-06-17:
If your project allows for installing librealsense from source code instead of a DKMS package, I recommend installing from source code using the "backend" method, which requires an internet connection but is not dependent on Linux versions or kernel versions, and does not require patches.

1.  Download the source code zip file for 2.33.1 from the 'Assets' list at the bottom of the page at the link below:

https://github.com/IntelRealSense/librealsense/releases/tag/v2.33.1

[image]

2.  Unzip the folder and go to the 'librealsense' root directory.  Input the command below to create a 'build' folder and then navigate to it:

**mkdir build && cd build**

3.  Once you are within the build directory, run the CMake command below to install 2.33.1 without the need for further steps such as patching:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

noblegasses, 2020-06-17:
Thanks for your help, Unfortunately, when trying to run cmake from build I get the following error: `bash: /usr/local/bin/cmake: cannot execute binary file: Exec format error` any Idea why this could be happening and what I could do to fix it?

MartyG-RealSense, 2020-06-17:
I believe that this error typically occurs when trying to execute a binary file that has already been built.  So I am not sure why it would occur when compiling a folder of source-code.

May I confirm please that you downloaded the **source code** zip file and then extracted its contents, please?

[image]

noblegasses, 2020-06-17:
I have pulled in the source code again, it is 2.33.1. below is an image of the contents of the extracted folder. 
[image]

MartyG-RealSense, 2020-06-17:
The extracted folder contents look the same as mine.  Once you have extracted the folder like you have done above, the next step is to change directory into the **build** folder with the cd build command and then run the CMake instruction from there (not from the root 'librealsense' directory).

noblegasses, 2020-06-17:
that is what I have done
[image]

noblegasses, 2020-06-17:
are you running this on the same device? I think [this](https://askubuntu.com/questions/648555/bash-program-cannot-execute-binary-file-exec-format-error) ask ubuntu thread is saying I might be getting the error because the source code is written for a different architecture.
https://askubuntu.com/questions/648555/bash-program-cannot-execute-binary-file-exec-format-error

MartyG-RealSense, 2020-06-17:
It does not seem to be an error related to Librealsense.  It has been reported to occur when trying to compile 64-bit code on a 32-bit machine.  The specification of Nanopi NEO4 says that it uses a 32-bit OS called Lubuntu.

noblegasses, 2020-06-17:
is there a 32 bit version of librealsense I can install??

MartyG-RealSense, 2020-06-17:
May I confirm that you are using 32-bit Lubuntu on your Nanopi, please?

noblegasses, 2020-06-17:
I don't believe so, but if these are not the correct commands please let me know
[image]

MartyG-RealSense, 2020-06-17:
Okay, you are using Ubuntu 18.04 LTS.  Thank you.

I did some further research about Nanopi boards specifically, and the process for getting a RealSense camera working on them does not seem to be straightforward, unfortunately.  If you work your way through the discussion in the link below, it may provide some useful insights about what to try next.

https://github.com/IntelRealSense/librealsense/issues/3042

noblegasses, 2020-06-17:
thank you, I am unfamiliar with kernel level work, but I will give this an attempt

RealSenseSupport, 2020-06-30:
@noblegasses  Any update from your side? Looking forward to your reply. Thanks!

dorodnic, 2020-07-09:
In my experience [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md) is the most reliable method for getting RealSense up and running on NanoPi.
I recommend getting a clean OS SD-Card and following the instructions. Feel free to open new issue if you encounter any realsense specific problems

closed 2020-07-09 by dorodnic