realsenseai/librealsense #6624: python application crashed when import pyrealsense2 on Win7

liukelinlin, 2020-06-17:
The device is a D415(Driver verision: 2.30.0.0), and installed SDK2.0 (Intel RealSense SDK - win7 - 2.35.2.1897). The viewer works as expected.
 
And installed python3.6.10 by miniconda, and "pip install pyrealsense2" successfully.
 
when i tried to "import pyrealsense2", it crashed. The error is attached.
https://support.intelrealsense.com/hc/user_images/JEkOxjDK-7M-hQ0fmg6Zng.jpeg

thanks,

RealSenseSupport, 2020-06-17:
@liukelinlin, could you please try to build the pyrealsense2 from source for Python3 for Win7? 
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#windows

Building the SDK2 for Win7 on either WIndows 7 or Windows 10 environment:
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_win7.md#building-from-source

Thank you

RealSenseCustomerSupport, 2020-06-19:
----------------------------------------------

Please let us know if further assistance is needed. Thank you.

The ticket will be closed in 7 days if there is no other question on the same topic.

closed 2020-06-26 by RealSenseSupport