realsenseai/librealsense #6626: RealsenseViewer.exe issues

lg93lggt, 2020-06-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400(435i) }                                          | 
| Firmware Version                   | (05.12.03.00)                                           | 
| Operating System & Version |   {Win 10                                                 | 
| Kernel Version (Linux Only)    |  (-)                                                           | 
| Platform                                 | PC                                                           |
| SDK Version                            |  {2.34.0.1470 }                                         | 
| Language                               |  {python }                                                | 
| Segment			         |  {others }                                                | 

### Issue Description
Here're 2 Issues:

1. When I use RealsenseViewer.exe , an error happened to like these images:
[image]
[image]
How to fix it?

2. When I record data, the images are blurry due to motion or shaking. How can I avoid it to happen? Turn down the exposure time by python SDK?

lg93lggt, 2020-06-18:
By the way, the stereo module and motion module run normally, if all 3 modules are activated .

MartyG-RealSense, 2020-06-18:
1.  The red timestamp indicates that the SDK is falling back on a software timestamp called System Time because it is unable to access hardware timestamps in the camera "metadata" information.  Documentation about metadata can be found in the link below.

https://dev.intelrealsense.com/docs/frame-metadata

Sometimes when the Viewer is launched there will be a pop-up box asking if you want to enable metadata.  If you do not receive this message, I have found that a way to trigger it is to unplug the camera and plug it into another USB port.

2.  The IR sensors of the D435i have a fast "global shutter" that can track very fast motion (even a vehicle traveling at full speed).  The RGB sensor has a slower "rolling shutter" though.

Would it be possible to try setting the RGB's FPS to 60 and see if that improves blur on the color image please?

Rolling shutter RGB image blur is discussed in the link below.

https://github.com/IntelRealSense/librealsense/issues/3554

lg93lggt, 2020-06-19:
1. Canot be sovled by unpluging the camera and plug it into another USB port.
2.Turning up  FPS to 60 do have a better progress.

MartyG-RealSense, 2020-06-19:
1.  The action of putting the camera in a different USB port does not on its own enable metadata.  The aim of doing it is to get the pop-up box about enabling metadata to appear on screen so that you can click the enable button on the pop-up.

Before the easy pop-up box system for enabling metadata on Windows was added to the RealSense Viewer, there was a more complicated method that involves using the **registry** interface of Windows.  If you do not have experience of registry editing then you should avoid it though, as mistakes made in altering the registry can break Windows.

https://dev.intelrealsense.com/docs/compiling-librealsense-for-windows-guide#section-enabling-metadata-on-windows

2.  Good to hear that using 60 FPS had a positive effect for you.

MartyG-RealSense, 2020-07-05:
Hi @lg93lggt   Do you still require assistance with this case please, or can it be closed?   Thanks!

closed 2020-07-18 by MartyG-RealSense