realsenseai/librealsense #6627: dkms not working with Ubuntu kernel 5.3.0-59-generic #53~18.04.1-Ubuntu

vii, 2020-06-18:
```sh
sudo modprobe videodev 
modprobe: ERROR: could not insert 'videodev': Unknown symbol in module, or unknown parameter (see dmesg)
```

```
[  387.514924] videodev: Unknown symbol media_devnode_remove (err -2)
[  387.514936] videodev: Unknown symbol media_request_object_find (err -2)
[  387.514948] videodev: Unknown symbol media_request_object_init (err -2)
[  387.514980] videodev: Unknown symbol media_request_object_unbind (err -2)
[  387.514992] videodev: Unknown symbol media_request_object_complete (err -2)
[  387.515021] videodev: Unknown symbol media_create_intf_link (err -2)
[  387.515028] videodev: Unknown symbol media_create_pad_link (err -2)
[  387.515051] videodev: Unknown symbol media_request_object_put (err -2)
[  387.515061] videodev: Unknown symbol media_device_register_entity (err -2)
[  387.515067] videodev: Unknown symbol media_request_get_by_fd (err -2)
[  387.515087] videodev: Unknown symbol media_request_put (err -2)
[  387.515097] videodev: Unknown symbol media_request_object_bind (err -2)
[  387.515107] videodev: Unknown symbol media_graph_walk_next (err -2)
[  387.515131] videodev: Unknown symbol media_get_pad_index (err -2)
[  387.515148] videodev: Unknown symbol media_graph_walk_start (err -2)
[  387.515154] videodev: Unknown symbol media_devnode_create (err -2)
[  387.515163] videodev: Unknown symbol media_device_unregister_entity (err -2)
```

```
$ dkms status
librealsense2-dkms, 1.3.14, 5.3.0-59-generic, x86_64: installed
r8168, 8.048.00, 5.3.0-59-generic, x86_64: installed
```

```
Package: librealsense2-dkms
Source: librealsense2
Version: 1.3.0-0ubuntu2
Architecture: all
Maintainer: Evgeni Raikhel <evgeni.raikhel@intel.com>
Installed-Size: 40507
Depends: dkms (>= 2.1.0.0)
Recommends: librealsense2-udev-rules
Section: kernel
Priority: extra
Homepage: https://github.com/IntelRealSense/librealsense
Description: Modified kernel modules for librealsense2
 ```

this seems similar to https://github.com/IntelRealSense/librealsense/issues/2806 which is for a different version

ev-mp, 2020-06-18:
@vii hello,
I'm using similar configuration and can't find any inconsistencies - the ko modukes are loaded as expected. Is it possible that you've replaced that file with non-dkms version?

>user$ **dkms status**
**_librealsense2-dkms, 1.3.14, 5.3.0-59-generic, x86_64: installed_**
nvidia, 450.36.06, 4.15.0-107-generic, x86_64: installed
nvidia, 450.36.06, 5.3.0-56-generic, x86_64: installed
nvidia, 450.36.06, 5.3.0-59-generic, x86_64: installed
nvidia, 450.36.06, 5.4.0-050400-generic, x86_64: installed
nvidia, 450.36.06, 5.4.0-37-generic, x86_64: installed
nvidia, 450.36.06, 5.4.0-38-generic, x86_64: installed
user$ **sudo modprobe videodev -v**
insmod /lib/modules/5.3.0-59-generic/kernel/drivers/media/mc/mc.ko 
**_insmod /lib/modules/5.3.0-59-generic/updates/dkms/videodev.ko_**
user$ **modinfo videodev**
**_filename:       /lib/modules/5.3.0-59-generic/updates/dkms/videodev.ko_**
**_version:        realsense2-dkms_**
alias:          char-major-81-*
license:        GPL
description:    Device registrar for Video4Linux drivers v2
author:         Alan Cox, Mauro Carvalho Chehab <mchehab@kernel.org>
srcversion:     0B91D32005B5FCAE57BF11E
depends:        mc
retpoline:      Y
name:           videodev
vermagic:       5.3.0-59-generic SMP mod_unload 
signat:         PKCS#7
signer:         
sig_key:        
sig_hashalgo:   md4

vii, 2020-06-19:
Problem: videodev requires following modules v4l2_common,videobuf2_v4l2,uvcvideo,videobuf2_common but they could not be loaded, as they were not present on this machine.

Solution:
```
$ sudo apt install linux-modules-extra-$(uname -r)
```

Sorry for the confusing report! Super excited for this to hit mainline!

Log of actions/notes:

The only videodev is built from the librealsense dkms
```
 sudo modprobe videodev -v
insmod /lib/modules/5.3.0-59-generic/updates/dkms/videodev.ko 
modprobe: ERROR: could not insert 'videodev': Unknown symbol in module, or unknown parameter (see dmesg)
```
I did an uninstall, then a dkms build, where it successfully patched and built, then dkms install and the same error came back.

Then, I tried with another kernel `Linux hostname 5.3.0-53-generic #47~18.04.1-Ubuntu SMP Thu May 7 13:10:50 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

```sh
$ sudo dkms install librealsense2-dkms/1.3.14
Kernel preparation unnecessary for this kernel.  Skipping...
applying patch 75-realsense-camera-formats-5.3.patch...patching file 5.3.0/drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 214 (offset -5 lines).
patching file 5.3.0/drivers/media/usb/uvc/uvcvideo.h
patching file 5.3.0/drivers/media/v4l2-core/v4l2-ioctl.c
Hunk #1 succeeded at 1308 (offset 6 lines).
patching file 5.3.0/include-overrides/uapi/linux/videodev2.h
Hunk #1 succeeded at 727 (offset 35 lines).

applying patch 76-realsense-metadata-5.3.patch...patching file 5.3.0/drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 2965 (offset 34 lines).
patching file 5.3.0/drivers/media/usb/uvc/uvcvideo.h

applying patch 77-realsense-hid-5.3.patch...patching file 5.3.0/drivers/iio/accel/hid-sensor-accel-3d.c
Hunk #1 succeeded at 276 (offset -13 lines).
Hunk #2 succeeded at 477 (offset -13 lines).
patching file 5.3.0/drivers/iio/gyro/hid-sensor-gyro-3d.c
Hunk #1 succeeded at 29 (offset -13 lines).
Hunk #2 succeeded at 76 (offset -13 lines).
Hunk #3 succeeded at 184 (offset -13 lines).
Hunk #4 succeeded at 200 (offset -13 lines).
Hunk #5 succeeded at 233 (offset -13 lines).
Hunk #6 succeeded at 423 (offset -13 lines).

applying patch 78-realsense-powerlinefrequency-control-fix-5.3.patch...patching file 5.3.0/drivers/media/usb/uvc/uvc_ctrl.c
Hunk #1 succeeded at 353 with fuzz 1 (offset -5 lines).

applying patch 79-realsense-makefile-adaptation-5.3.patch...patching file 5.3.0/drivers/media/usb/uvc/Makefile
patching file 5.3.0/drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 207 (offset 37 lines).
patching file 5.3.0/drivers/media/v4l2-core/Makefile
patching file 5.3.0/drivers/media/v4l2-core/v4l2-dev.c

Building module:
cleaning build area...
if [ -d /var/lib/dkms/librealsense2-dkms/1.3.14/build/drivers ]; then rmdir /var/lib/dkms/librealsense2-dkms/1.3.14/build/drivers; fi && make -C /lib/modules/5.3.0-53-generic/build M=/var/lib/dkms/librealsense2-dkms/1.3.14/build/5.3.0/drivers/media/usb/uvc uvcvideo.ko DRIVER_VERSION_SUFFIX='.realsense-1.3.14' && mkdir -p /var/lib/dkms/librealsense2-dkms/1.3.14/build/drivers/media/usb/uvc/ && cp /var/lib/dkms/librealsense2-dkms/1.3.14/build/5.3.0/drivers/media/usb/uvc/uvcvideo.ko /var/lib/dkms/librealsense2-dkms/1.3.14/build/drivers/media/usb/uvc/ && make -C /lib/modules/5.3.0-53-generic/build M=/var/lib/dkms/librealsense2-dkms/1.3.14/build/5.3.0/drivers/media/v4l2-core videodev.ko && mkdir -p /var/lib/dkms/librealsense2-dkms/1.3.14/build/drivers/media/v4l2-core/ && cp /var/lib/dkms/librealsense2-dkms/1.3.14/build/5.3.0/drivers/media/v4l2-core/videodev.ko /var/lib/dkms/librealsense2-dkms/1.3.14/build/drivers/media/v4l2-core/ && make -C /lib/modules/5.3.0-53-generic/build M=/var/lib/dkms/librealsense2-dkms/1.3.14/build/5.3.0/drivers/iio/accel hid-sensor-accel-3d.ko && mkdir -p /var/lib/dkms/librealsense2-dkms/1.3.14/build/drivers/iio/accel && cp /var/lib/dkms/librealsense2-dkms/1.3.14/build/5.3.0/drivers/iio/accel/hid-sensor-accel-3d.ko /var/lib/dkms/librealsense2-dkms/1.3.14/build/drivers/iio/accel/ && make -C /lib/modules/5.3.0-53-generic/build M=/var/lib/dkms/librealsense2-dkms/1.3.14/build/5.3.0/drivers/iio/gyro hid-sensor-gyro-3d.ko && mkdir -p /var/lib/dkms/librealsense2-dkms/1.3.14/build/drivers/iio/gyro && cp /var/lib/dkms/librealsense2-dkms/1.3.14/build/5.3.0/drivers/iio/gyro/hid-sensor-gyro-3d.ko /var/lib/dkms/librealsense2-dkms/1.3.14/build/drivers/iio/gyro/.......
Signing module:
 - /var/lib/dkms/librealsense2-dkms/1.3.14/5.3.0-53-generic/x86_64/module/uvcvideo.ko
 - /var/lib/dkms/librealsense2-dkms/1.3.14/5.3.0-53-generic/x86_64/module/videodev.ko
 - /var/lib/dkms/librealsense2-dkms/1.3.14/5.3.0-53-generic/x86_64/module/hid-sensor-accel-3d.ko
 - /var/lib/dkms/librealsense2-dkms/1.3.14/5.3.0-53-generic/x86_64/module/hid-sensor-gyro-3d.ko
Secure Boot not enabled on this system.
cleaning build area...

DKMS: build completed.

uvcvideo.ko:
[... 27 lines cut ...]
depmod...

DKMS: install completed.
```

but the same problem -
```
  106.973624] videodev: Unknown symbol media_devnode_remove (err -2)
[  106.973633] videodev: Unknown symbol media_request_object_find (err -2)
[  106.973643] videodev: Unknown symbol media_request_object_init (err -2)
[  106.973674] videodev: Unknown symbol media_request_object_unbind (err -2)
[  106.973684] videodev: Unknown symbol media_request_object_complete (err -2)
[  106.973708] videodev: Unknown symbol media_create_intf_link (err -2)
[  106.973713] videodev: Unknown symbol media_create_pad_link (err -2)
[... cut ...]
[  106.973759] videodev: Unknown symbol media_request_put (err -2)
[  106.973767] videodev: Unknown symbol media_request_object_bind (err -2)
[  106.973774] videodev: Unknown symbol media_graph_walk_next (err -2)
[  106.973792] videodev: Unknown symbol media_get_pad_index (err -2)
[  106.973806] videodev: Unknown symbol media_graph_walk_start (err -2)
[  106.973812] videodev: Unknown symbol media_devnode_create (err -2)
[  106.973820] videodev: Unknown symbol media_device_unregister_entity (err -2)
````

ev-mp, 2020-06-20:
@vii, the standard kernel modules should be obtained by either installing the default Ubuntu desktop distribution or by using the Debian upgrade over the original LTS distro with "sudo apt dist-upgrade"

In case you use a modified kernel installation then you'll need to install the missing drivers package regardless of DKMS, as those are required by the Depth camera for the basic functionality.
In general, for every LTS kernel installed the following packages should be present:
linux-image-$(uname -r)
linux-headers-$(uname -r)
linux-modules-$(uname -r)
linux-modules-extra-$(uname -r)

RealSenseSupport, 2020-07-01:
@vii Did you get chance to try as ev-mp suggested? Looking forward to your update. Thanks!

ev-mp, 2020-07-06:
Closed due to no customer feedback.

closed 2020-07-06 by ev-mp