realsenseai/librealsense #6630: Hello everyone! How can I change the frame rate of D435 to 4FPS for depth frame using python? Now, I can change to just 6FPS. Also I would be glad if you recommend me how to change the depth units. For example, I got the depth value of 3500 in CSV instead of 3.500 meters.

ngaye1511, 2020-06-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-06-18:
A feature called **Genlock** was recently introduced in SDK version 2.35.2 that allows custom FPS speeds to be used, such as very slow ones.  This enables frequencies that were previously not supported to be achieved.

https://www.intelrealsense.com/depth-camera-external-sync-trigger/

The link below has information on changing the depth units in Python:

https://support.intelrealsense.com/hc/en-us/community/posts/360034848973/comments/360008694314

ngaye1511, 2020-06-23:
Thank you for your suggestions.

closed 2020-06-23 by ngaye1511