realsenseai/librealsense #6631: Where to find the connector for T261 module ?

PLMR-FR, 2020-06-18:
Hello,

It seems to be very difficult to find the I-pex NOVASTACK connector (20708-034E-1) in low quantity in order to use the T261 module.

I contacted I-pex but they appear to not be interested to sell low quantity and they stop to answers for my request.

Is there any place in Europe to find this connector for low quantities ? (10 pcs) We are prototyping and we can't by a reel of 5000 pcs.
I checked on mouser, digikey, rs but not available.

Thanks for your help.

PLMR

dorodnic, 2020-07-09:
I don't think we can recommend any reseller for this part (not sure if we have such information)

closed 2020-07-09 by dorodnic

skylouis, 2020-09-10:
Hi @PLMR-FR, apparently you can ask I-pex for free sample of this connector.