realsenseai/librealsense #6635: Hardware Reset for D435 After Flashing Invalid File For Firmware

matthewha123, 2020-06-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |Attempt at Flashing 05.12.05.00 | 
| Operating System & Version |   Ubuntu 18.0.4 | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                        | 
| Platform                                 |  Nvidia Jetson Nano  |
| SDK Version                            |  2.32.1                       | 
| Language                            |  C++                        | 
| Segment			|  others                    | 

### Issue Description
I was running the `rs-fw-update` tool and I accidentally passed in the zip file for the firmware, rather than the .bin file in the unzipped directory. 

Now `rs-enumerate-devices` gives this output:
`No device detected. Is it plugged in?`

And running the tool in recovery mode: `rs-fw-update -r -f <firmware.bin>` gives this output and returns immediately without installing anything.
`update to FW: Signed_Image_UVC_5_12_5_0.bin`

How can I recover this camera?

dorodnic, 2020-06-18:
Hi @matthewha123 
Try unplugging and re-plugging the camera, check if it is visible in lsusb and realsense-viewer and check it is connected via USB3. 
It should either re-enumerate as D435 or appear as D435 recovery device with the option to burn recommended firmware.

RealSenseSupport, 2020-06-22:
Hi @matthewha123 Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

matthewha123, 2020-06-22:
Thank you that worked!

closed 2020-06-22 by matthewha123