realsenseai/librealsense #6636: L515 configurable depth units

bmegli, 2020-06-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515  | 
| Firmware Version                   | 01.04.01.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-43                                         | 
| Platform                                 |  PC  |
| SDK Version                            |  2.35.2                        | 

`

### Issue Description

From what I see in `realsense-viewer` L515 only supports depth units 0.00025.
Also, querying device from code returns the range [0.00025, 00025] for depth units.

Are there any plans to make depth units configurable for L515 like it is possible for D400 series?

### Why would I want it

I am hardware encoding depth streams to HEVC Main10, using 10 bit P010LE format. This means I am losing 6 bits of precision. For D400 series it is possible to control encoding range-precision trade-off with depth units. This means that host device doesn't do any processing of the data, just passes it to hardware encoder. I would like to keep similar pipeline for L515 if possible.

RealSenseSupport, 2020-06-26:
Hi @bmegli 

At this time we do not have this capability.  This would be a new feature enhancement.  We will take note but no timetable as to when we would be able to investigate an implementation of this feature.

bmegli, 2020-06-29:
Ok, thank you.

L515 doesn't support advanced mode like D400 so I have to do other post-processing on host device anyway.

This means that I may just as well simulate depth units behavior, I am closing the issue, thanks.

closed 2020-06-29 by bmegli