realsenseai/librealsense #6637: L515 device side depth clamping

bmegli, 2020-06-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515  | 
| Firmware Version                   | 01.04.01.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-43                                         | 
| Platform                                 |  PC  |
| SDK Version                            |  2.35.2                        | 

`

### Issue Description

From what I see in `realsense-viewer` L515 doesn't support device side depth clamping and in general advanced mode similar to D400 series.

Are there any plans for L515 advanced mode similar to D400 series?

dorodnic, 2020-06-22:
Hi @bmegli 
Could you clarify what is side-depth clamping?

bmegli, 2020-06-22:
Hi @dorodnic,

Sorry for not being specific.

Precisely I mean device side advanced mode that allows shaping the data on the device, e.g:

```C++
  rs400::advanced_mode advanced = profile.get_device();
  pipe.stop(); //workaround the problem with setting advanced_mode on running stream
  STDepthTableControl depth_table = advanced.get_depth_table();
  depth_table.depthClampMax = P010LE_MAX;
  advanced.set_depth_table(depth_table);
  profile = pipe.start(cfg);
```

For example the above code results in device returning depth above P010LE_MAX as 0.

Naturally similar effect may be achieved on the host side by post processing the data.

So the question is:

Is something similar to advanced mode of rs400 planned for L515 or should we just perform host side post processing?

dorodnic, 2020-06-22:
For now, post-processing is the way to go. There is `rs2::threshold` in the SDK, but I understand this is not your question.
At the moment, everything that camera firmware allows you to do is exposed in the SDK / Viewer. What you requesting sounds possible within ASIC algo pipe, but it would require new firmware. 
We continue to add firmware and software features to our stereo cameras, mostly driven by customer feedback. I cannot speak on behalf of the organisation, but it seems to be the plan for Lidar family of cameras as well.

bmegli, 2020-06-23:
Ok, thanks!

closed 2020-06-23 by bmegli