realsenseai/librealsense #6638: Issues connecting D435 and T265 together using Python

bobgates, 2020-06-18:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 and D435i } | 
| Firmware Version                   | D435: 05.12.05.00 T265: 0.2.0.951 | 
| Operating System & Version |  Linux (Ubuntu 20.4)  | 
| Kernel Version (Linux Only)    |   5.4.0-1012-raspi                                       | 
| Platform                                 | Raspberry P 3B  |
| SDK Version                            | 2.35.2                         | 
| Language                            |  python                         | 
| Segment			|  {Robot }                          | 

### Can't get T265 and D435i working at the same time on Raspberry Pi 3B in Python

I have not been able to figure out how to get the T265 and D435i working at the same time on a Raspberry Pi 3B. My test code is below. If I set either use265 or use435 to true, all is well, but if I set them both to true, then the code fails where it is waiting for the T265 to deliver a frame:
'''
    tframes = pipeline_2.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000
'''

I've followed the advice I could find online, but a lot of it is either old, or not exactly what I'm after (#6362, https://github.com/keijiro/Rsvfx/issues/11, #5463, which points to #3697, which didn't work for me). The code here is derived from #6184, which describes the same issue, more-or-less, under Windows.

I battled a lot with '''Could not open device failed to set power state''' errors, related to things going wrong, and only pulling out USB connectors and re-inserting them would get me past that point. The code below doesn't provoke those errors, but it doesn't work. 

I've seen advice on using one context for both pipelines (#5509, done) and making sure to start the T265 first (done). #5614 comes pretty close, and describes much of what is making me tear my hair out. I also see a set of errors when using '''rs-enumerate-devices -d''' with the T265 and D435i both connected: '''(handle-libusb.h:95) failed to claim usb interface:''' and '''Unable to open device interface'''. 

Is the T265/D435i combination viable? Should I move away from Pi? If so, to what, for a low cost toy robot car? Getting a T265 to work seems surrounded by a good deal of randomness and is affecting lots of people with lots of mixes of OS in lots of different ways. neilyoung expresses my frustration in #5756, and makes me think I should steer clear of the T265 for a while. But there's nothing else out there that makes GPS-denied navigation quite so easy.

I'd really appreciate any input.

'''
#!/usr/bin/python3

import pyrealsense2 as rs

use435 = True
use265 = False

context = rs.context()

if use265:
    pipeline_2 = rs.pipeline(context)
    config_2 = rs.config()
    config_2.enable_device('905312110841')
    config_2.enable_stream(rs.stream.pose)
    a=pipeline_2.start(config_2)
    print('T265 pipeline start result:',a)

if use435:
    pipeline_1 = rs.pipeline(context)
    config_1 = rs.config()
    config_1.enable_device('833612071395')
    b=pipeline_1.start(config_1)
    print('D435 pipeline start result:',b)

for i in range(500):
    if use435:
        dframes = pipeline_1.wait_for_frames()
        depth = dframes.get_depth_frame()
        if not depth: 
            continue
        dist = depth.get_distance(320,240)
        print("{:.4} ".format(dist), end='')

    if use265:
        tframes = pipeline_2.wait_for_frames()
        pose = tframes.get_pose_frame()
        if pose:
            data = pose.get_pose_data()
            print("{} ".format(pose.frame_number), end='')

    if i % 15 == 0:
        print()

print()

if use435:
    pipeline_1.stop()
if use265:
    pipeline_2.stop()
'''

dexterpengji, 2020-07-20:
Is there any update on this issue? I have the same problem on a different machine, it’s a X86 PC with Ubuntu 18.04.4.
I have 3 d435i and 1 t265, if I open the t265 first, the other three d435i won’t start their pipes. If start three d435i first, the t265 won’t start.
If I use  config.enable_device(serial_number), it even does not work without any other camera.
I will try some old versions of SDK to see if it worked before.(I’ll try 2.19 first) 
Since all cameras can work well at the same time in realsense viewer, there should be a way to let all cameras work in the python wrapper.

RealSenseSupport, 2020-10-26:
Hi,

Will you be needing further help with this? Are you still experiencing this on latest SDK/FW?

If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-10-30 by RealSenseSupport