realsenseai/librealsense #6641: Interference problem in multiple L515s

kojiohno, 2020-06-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   |01.04.01.00 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            | 2.35.2  | 
| Language                            |  C  | 
| Segment			|  others  | 

### Issue Description

When two L515s are started at the same time, it seems that they interfere with each other and the depth data is disturbed.
Hiding one of the cameras will reduce the disturbance.

Is it a problem that can be solved by external synchronization?
Where can I find information on the L515 external sync?
[image]

RealSenseSupport, 2020-06-19:
Hi @kojiohno 

This is expected based on the technology.  This is not the same as stereo pair in our D400 series cameras where the more texture on the scene the better.  For L515, the laser projector, the pattern and the processing of the pattern when the beam is reflected back to the IR photodiode is a very specific process.  So having two L515 projecting where the lasers are overlapping will cause some interference with any given L515 in that scenario and that interference will cause depth data to suffer.  

With that being said, we are working on a sync functionality but it is not ready at this point in time.

I would say that what you're seeing now is expected and not a problem.  This is how the technology is.  You can continue to check back with the release notes of librealsense and/or datasheet (https://dev.intelrealsense.com/docs/lidar-camera-l515-datasheet), whitepapers (https://dev.intelrealsense.com/docs/whitepapers) regarding a sync function that will allow for synchronizing multiple L515 cameras together.

Is there anything else you need on this topic?

kojiohno, 2020-06-19:
Thank you for your answer.

I understand this is the expected behavior based on the technology. However, when the synchronization function becomes available, is it possible that interference will be mitigated by slightly shifting the IR laser irradiation timing of each device?
Also, do you have any idea when the sync function will be released?

RealSenseSupport, 2020-06-26:
Hi @kojiohno 

We are working on it and working on documentation as well.  We will update you when the information is ready.

xiebei, 2020-07-21:
Any updates? Can I turn off the emitter like D415?

RealSenseSupport, 2020-07-24:
@xiebei - We are still working on the documentation and details around this.  Thanks for checking in.  No ability to turn off the emitter on the L515.  The emitter is essential to the LiDAR technology in L515.

RealSenseSupport, 2020-08-13:
@kojiohno @xiebei and others on this thread

We've posted the following document that was designed specifically to discuss how to connect and configure for multiple L515 cameras

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

This document should be exactly what you are looking for.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-08-20 by RealSenseSupport