realsenseai/librealsense #6643: stops to "rs2::depth_frame depth = frames.get_depth_frame();" for a long time

DreamLFairy, 2020-06-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515
| Operating System & Version |  RaspberryPi 4B Raspbian (no ubuntu)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |Raspberry Pi                                           |
| SDK Version                            |  librealsense -2.35.2     master              | 
| Language                               | c++                                                         | 

### Issue Description
I run the example-"rs-hello-realsense.cpp", the process stops to "rs2::depth_frame depth = frames.get_depth_frame();"  and waits for a long time (about 2 minutes), and outputs "The camera is facing an object  0 meters away ".
And then loop the process above;
1. The first, I want to know why does this happen.
2. The sencond, how can I to solve it?
Thanks for your reply.

dorodnic, 2020-06-22:
Hi @DreamLFairy 

I ran some tests to reproduce the issue with Raspberry Pi, but both pointcloud and hello-realsense were functioning properly. 
I'm on Raspbian release from February, but it should equally apply to the rebranded Raspberry Pi OS. I tried using USB-C power as well as Power-over-Ethernet. Also tested with recommended firmware (1.4.1) and an earlier one. 

If camera was connected to USB2, applications would report "No device connected". If udev rules were not up to date, you would get error about ACCESS. So I'm not sure what is going on. hello-realsense may report 0 if there's a lot of ambient light, but I have no explanation for pointcloud behaviour. 

Is this a newly shipped L515 or a pre-production device?

RealSenseSupport, 2020-06-29:
Hi @DreamLFairy 

Were you able to have a look at the items proposed by dorodnic?

DreamLFairy, 2020-06-30:
I've just solved the problem. I used a 32-bit Raspbian system before. Now I've switched to a 64 bit Raspbian system, it works very well.

RealSenseSupport, 2020-06-30:
Glad it's working. We'll go ahead and close this issue but should you have further issues feel free to re-open or create a new one.

Thanks

closed 2020-06-30 by RealSenseSupport