realsenseai/librealsense #6649: Qt  and Intel realsense Getting started

ZiSo89, 2020-06-19:
| | |
|---|---|
|**librealsense**|2.33.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|943222071442|
|**Firmware Version**|05.12.03.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|943623022438|
|**Firmware Update Id**|943623022438|

Firstly there is not technical issue, I didn't know where to ask as a beginner.
So If you can help...
I want to create a project using:

- **Qt** 5.15.0
- **Visual studio** 2019
- C++
- openCV 4.2.0 (if is necessary)

Actually to start I need a simpler project like those from the Examples for Intel RealSense SDK 2.0,
the "**im-show**"
or/and the "hello-realsense"

A _code example_ will be the best or even better a tutorial.
Till now I didn't found something on google. 
In case the above is difficult an advice on how to get start will be fine. 

For example, I found: "[Qt and openCV on Windows with MSVC](https://wiki.qt.io/How_to_setup_Qt_and_openCV_on_Windows_with_MSVC2017)" 
Is it a good place for me to start?

Thank you in advance.

MartyG-RealSense, 2020-06-19:
Hi @ZiSo89   I researched your request very carefully and could not find one central guide that explains everything about linking QT to Librealsense on  Windows.  I have tried to assemble together a starting-point of information though.

The first step is to install the Librealsense SDK, which you may have done by now.

Once the SDK is installed then you can go into QT Creator and follow these steps:

1.  Go to the **Open File / Projects** menu option.
2.  Browse for the **CMakeLists.txt** file at this location: librealsense/CMakeLists.txt

ZiSo89, 2020-06-19:
I'm going somewhere with my own question, :)
I found online and combined few projects with Qt and openCV and I rewrite in Visual Studio.

[image]

Except of few bug and safety functionality, is running. I think in next days I will fix those problems, and then I will upload it on GitHub

MartyG-RealSense, 2020-06-19:
I'm glad you were able to make quick progress.  Good luck with your further tests!

ZiSo89, 2020-06-20:
Maybe you can have a look and suggest for improvements. <br> https://github.com/ZiSo89/QtIntelRealSense

MartyG-RealSense, 2020-06-20:
My computer is not set up to test programs unfortunately.  Other RealSense QT users will find your program useful though.  Thanks so much for sharing it!

closed 2020-06-20 by ZiSo89