realsenseai/librealsense #6650: Doesn't work with Linux, only Ubuntu Linux.

andrewhodel, 2020-06-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I've replaced the instances of generic with my kernel version for linux-headers in scripts/patch-realsense-ubuntu-lts.sh and it still reports Unsupported kernel version 4.19.0-8-amd64 . The patches are maintained for Ubuntu16 (Xenial) with LTS kernel versions 4.4, 4.8, 4.10, 4.13 and 4.15

Is there some reason for this, not supporting Linux and only supporting Ubuntu Linux?

I hate to install a distribution of Linux based on the one I normally use, Debian.

andrewhodel, 2020-06-19:
Dug deeper in #6651 will create a patch for Ubuntu once patch for 4.16.1 is fixed.

closed 2020-06-19 by andrewhodel

MartyG-RealSense, 2020-06-19:
Hi @andrewhodel   I see that you closed the case, but I will share the answer that I was working on anyway.  :)

There is a "backend" installation method that requires an internet connection but is not dependent on Linux versions or kernel versions, and does not require patching.

First, download the source code of an SDK version by going to the 'Releases' page and downloading the 'Source code' zip file from the "Assets" file list at the bottom of each version's details.

https://github.com/IntelRealSense/librealsense/releases/

[image]

Once the source code zip for the SDK version that you want to install has been extracted from the ip folder, follow the instructions in the link below to build Librealsense without dependence on Linux, kernels or patches.

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

andrewhodel, 2020-06-19:
> #6368 (comment)

https://github.com/IntelRealSense/librealsense/issues/6652

andrewhodel, 2020-06-19:
I just don't understand why your instructions on Github and on the Intel
SDK site are so terrible.

Why don't you just tell people to rebuild the kernel, apply the patches
needed and use the kernel they built?

Then it works with any Linux distribution or version that supports the
library versions (so 4.18.1+ or whatever).

Andrew

andrewhodel, 2020-06-19:
You say this, Once the source code zip for the SDK version that you want to
install has been extracted from the ip folder, follow the instructions in
the link below to build Librealsense without dependence on Linux, kernels
or patches.

But this is not true.

The DKMS file for the kernel drivers that are patched (uvcvideo) don't
support Debian 10 for example, the kernel version is excluded in DKMS.conf

If you told people to just build a new kernel, there's no problem running
4.18 or 5.0+ on Debian 10

Your instructions are too high, they need to get back to understanding what
a kernel module is.

Andrew

MartyG-RealSense, 2020-06-19:
Yes, the process that I specified is designed for source code installation.  DKMS packages have their own installation process.

I do understand your frustration.  The instructions try to meet the needs of as wide an audience as possible on multiple types of computer / computing device, though there are cases where further support may be required to successfully complete an installation.

andrewhodel, 2020-06-19:
You are giving a wide audience instructions for effectively a single
platform, Ubuntu LTS.

Why don't you document how to do this the right way?

Andrew

MartyG-RealSense, 2020-06-19:
Ubuntu is the most widely used platform, though we do provide options for other forms of Linux and create specialist solutions for platforms such as Raspberry Pi and Nvidia Jetson.

I cannot speak for the creators of the documentation, but I can assure that they try to do their best for everyone and give individual specialist support on the forum where necessary.

Thanks for your patience, it really is appreciated!

andrewhodel, 2020-06-19:
> Ubuntu is the most widely used Linux platform, though we do provide

Can you at least list the patches in the scripts/ directory by kernel
versions, not distribution specific ones?

Andrew

MartyG-RealSense, 2020-06-19:
I am on the Support team and are not involved with development decisions, so I cannot comment on that point.  I recommend raising the points in your open case.  Thanks.

ev-mp, 2020-06-20:
@andrewhodel hello,
I've followed the [PR you submitted](https://github.com/IntelRealSense/librealsense/pull/6653) as I wasn't sure about its purpose or to whom it was addressed.
Here it seem to be the proper context for reply -

>I've replaced the instances of generic with my kernel version for linux-headers in scripts/patch-realsense-ubuntu-lts.sh and it still reports Unsupported kernel version 4.19.0-8-amd64 . The patches are maintained for Ubuntu16 (Xenial) with LTS kernel versions 4.4, 4.8, 4.10, 4.13 and 4.15
Is there some reason for this, not supporting Linux and only supporting Ubuntu Linux?

That is not correct. Librealsense2 SDK and RealSense Depth cameras and is in no way limited for Ubuntu.They can and be deployed on any Linux distribution and platform that satisfy the following minimal requirements: Kernel 4.4+/ GCC ~4.8 and USB2/3 port according to the camera model.
Most of the camera and SDK functions become available immediately upon installation without patches  - Device enumeration, controls, data streaming.

The kernel patches are intended to enable advanced capabilities unique to Realsense devices, and also to fill kernel gaps for functionalities that are yet-to-be implemented or buggy. Still, the core feature that justifies the existence of those patches is the [frame metadata](https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md).
As for the selected distribution - we maintain patches for Ubuntu LTS kernels in a form of script and also as a precompiled DKMS module. Both methods are functionally-equivalent but for obvious reasons we recommend DKMS as the preferred installation method ([Documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#linux-distribution)). 
In addition to the patches provided by Intel for Ubuntu there are multiple patches that were generated and  provided by community. Unfortunately those tend to become obsolete as in most cases community members do not update them after the initial submissions.
And this is the core distinction - the officially-"maintained" patches are being updated continuously and reflect the up-to-date state of all the supported devices/required functionalities. So when the need arise they are used as a baseline for a new port by maintainers or community.

>I just don't understand why your instructions on Github and on the Intel
SDK site are so terrible.

The online documentation is being checked and updated periodically. Can you provide specific examples?

>Why don't you just tell people to rebuild the kernel, apply the patches
needed and use the kernel they built?
Then it works with any Linux distribution or version that supports the
library versions (so 4.18.1+ or whatever).

Because not only it is not precise, It is actually the is opposite of the message we convey:
- We do not require the users to apply kernel patches. The users are free to experiment with the default settings, and for many users this is sufficient. 
Ubuntu users have a certain advantage as they have the kernel patches/installation package ready for them already. But, as pointed above by @MartyG-RealSense , for users that don't use Ubuntu 
 the SDK provides a "driver-less" build. This configuration delivers all the "extra" functionality (metadata) without touching the kernel. Currently this is the recommended method for RaspberryPi and Tegra boards. Additionally we are open for collaboration and ready to provide technical assistance for users that port the patches to other kernels/distros.
- For most* distributions it is unnecessary - the patches apply to several loadable modules so it is technically inefficient to rebuild the whole kernel to replace those 6-7  `*.ko` modules.
**There are several exceptions for this case, such as Tegra or Yocto distros, but they also have dedicated handling/installation guidance.

andrewhodel, 2020-06-20:
They have to apply the kernel patches to get it to work on distributions
other than Ubuntu.

I built all the scripts for linux vanilla and debian 10 and submitted a
pull request to the github project for realsense with them.

https://github.com/IntelRealSense/librealsense/pull/6653

Andrew