realsenseai/librealsense #6655: Could not build librealsense 2.35.2 on Jetson

atinfinity, 2020-06-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |   | 
| Operating System & Version |   Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                        | 
| Platform                                 |  NVIDIA Jetson TX2  |
| SDK Version                            |  2.35.2                           | 
| Language                            |  C++                         | 

### Issue Description
I tried to build librealsense 2.35.2 on Jetson.
But, I could not build librealsense.

#### Build Instruction
```shell
$ mkdir build
$ cd build
$ cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true ..
$ make -j4
```

#### Error Message
```shell
[ 24%] Building CXX object CMakeFiles/realsense2.dir/src/mock/sql.cpp.o
[ 24%] Building CXX object CMakeFiles/realsense2.dir/src/mock/recorder.cpp.o
[ 25%] Building CUDA object CMakeFiles/realsense2.dir/src/proc/cuda/cuda-align.cu.o
[ 25%] Building CXX object CMakeFiles/realsense2.dir/src/proc/cuda/cuda-pointcloud.cpp.o
In file included from /home/dandelion/librealsense/src/proc/cuda/cuda-pointcloud.h:5:0,
                 from /home/dandelion/librealsense/src/proc/cuda/cuda-pointcloud.cpp:3:
/home/dandelion/librealsense/src/proc/cuda/../pointcloud.h:11:40: error: expected initializer before ‘:’ token
     class LRS_EXTENSION_API pointcloud : public stream_filter_processing_block
                                        ^
In file included from /home/dandelion/librealsense/src/proc/cuda/cuda-pointcloud.cpp:3:0:
/home/dandelion/librealsense/src/proc/cuda/cuda-pointcloud.h:10:5: error: expected class-name before ‘{’ token
     {
     ^
/home/dandelion/librealsense/src/proc/cuda/cuda-pointcloud.h:14:15: error: ‘float3’ does not name a type; did you mean ‘float’?
         const float3 * depth_to_points(
               ^~~~~~
               float
/home/dandelion/librealsense/src/proc/cuda/cuda-pointcloud.cpp: In constructor ‘librealsense::pointcloud_cuda::pointcloud_cuda()’:
/home/dandelion/librealsense/src/proc/cuda/cuda-pointcloud.cpp:11:42: error: class ‘librealsense::pointcloud_cuda’ does not have any field named ‘pointcloud’
     pointcloud_cuda::pointcloud_cuda() : pointcloud("Pointcloud (CUDA)") {}
                                          ^~~~~~~~~~
/home/dandelion/librealsense/src/proc/cuda/cuda-pointcloud.cpp: At global scope:
/home/dandelion/librealsense/src/proc/cuda/cuda-pointcloud.cpp:17:26: error: no ‘const float3* librealsense::pointcloud_cuda::depth_to_points(rs2::points, const rs2_intrinsics&, const rs2::depth_frame&, float)’ member function declared in class ‘librealsense::pointcloud_cuda’
         float depth_scale)
                          ^
CMakeFiles/realsense2.dir/build.make:902: recipe for target 'CMakeFiles/realsense2.dir/src/proc/cuda/cuda-pointcloud.cpp.o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/proc/cuda/cuda-pointcloud.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/realsense2.dir/all' failed
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

MartyG-RealSense, 2020-06-20:
Hi @atinfinity  There is currently a bug in building Librealsense 2.35.2 for Jetson with -**DBUILD_WITH_CUDA=true**.  A fix is detailed at the link below.

https://github.com/IntelRealSense/librealsense/issues/6573#issuecomment-643725732

atinfinity, 2020-06-20:
@MartyG-RealSense Thank you for your reply.
I tried to fix refer to https://github.com/IntelRealSense/librealsense/pull/6581. As a result, I can build librealsense. So, I closed this issue.

closed 2020-06-20 by atinfinity