realsenseai/librealsense #6656: Could not detect L515 on Jetson TX2

atinfinity, 2020-06-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Operating System & Version |   Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                        | 
| Platform                                 |  NVIDIA Jetson TX2  |
| SDK Version                            |  2.35.2                           | 
| Language                            |  C++                         | 

### Issue Description
I tried to use L515 on Jetson TX2.
But, `rs-enumerate-devices` could not detect L515.

```shell
$ rs-enumerate-devices
No device detected. Is it plugged in?
```

So, I connected L435 to same Jetson TX2.
In this case, rs-enumerate-devices detects L435.

#### Build Instruction
```shell
$ mkdir build
$ cd build
$ cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true ..
$ make -j4
```

ev-mp, 2020-06-20:
@atinfinity hello, did you update the `udev-rules` during v2.35.2 installation - 
https://github.com/IntelRealSense/librealsense/blob/master/scripts/setup_udev_rules.sh ?

atinfinity, 2020-06-21:
@ev-mp I launched `scripts/setup_udev_rules.sh` again. But, `rs-enumerate-devices` could not detect L515.

dorodnic, 2020-06-21:
Hi @atinfinity 
Maybe you already tried this - 
You write `make -j4` and then `rs-enumerate-devices`. 
What about `./tools/enumerate-devices/rs-enumerate-devices`?

atinfinity, 2020-06-21:
@dorodnic Thank you for your comment.

At first, I uninstalled older version by `sudo make uninstall`.
And, I removed source code.

Afterwards, I installed the latest version by the following command.

```shell
$ make -j4
$ sudo make install 
$ rs-enumerate-devices
```

ev-mp, 2020-06-22:
@atinfinity , additionally L515 is a USB3 device, so it could be that it is identified but only as USB2, which not sufficient to operate.
Try to disconnect and reconnect it fast and firmly to USB3 port.
Adding the kernel log when re-plugging the camera will also shed some light - `dmesg -wT`

atinfinity, 2020-06-22:
@ev-mp I disconnect and re-plugged L515 and I checked the result of `dmesg -wT`

```shell
[火  6月 23 02:16:25 2020] usb 2-1: new SuperSpeed USB device number 11 using tegra-xusb
[火  6月 23 02:16:25 2020] usb 2-1: New USB device found, idVendor=8086, idProduct=0b64
[火  6月 23 02:16:25 2020] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[火  6月 23 02:16:25 2020] usb 2-1: Product: Intel(R) RealSense(TM) 515
[火  6月 23 02:16:25 2020] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Camera
[火  6月 23 02:16:25 2020] usb 2-1: SerialNumber: 00000000F0172443
[火  6月 23 02:16:25 2020] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[火  6月 23 02:16:25 2020] uvcvideo: Unknown video format 20202043-0000-0010-8000-00aa00389b71
[火  6月 23 02:16:25 2020] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 515 (8086:0b64)
[火  6月 23 02:16:25 2020] uvcvideo: Unable to create debugfs 2-11 directory.
[火  6月 23 02:16:25 2020] uvcvideo: Unable to create debugfs 2-11 directory.
[火  6月 23 02:16:25 2020] uvcvideo 2-1:1.0: Entity type for entity Extension 3 was not initialized!
[火  6月 23 02:16:25 2020] uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
[火  6月 23 02:16:25 2020] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
[火  6月 23 02:16:25 2020] input: Intel(R) RealSense(TM) 515 as /devices/3530000.xhci/usb2/2-1/2-1:1.0/input/input12
[火  6月 23 02:16:25 2020] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 515 (8086:0b64)
[火  6月 23 02:16:25 2020] uvcvideo: Unable to create debugfs 2-11 directory.
[火  6月 23 02:16:25 2020] uvcvideo 2-1:1.4: Entity type for entity Processing 10 was not initialized!
[火  6月 23 02:16:25 2020] uvcvideo 2-1:1.4: Entity type for entity Extension 11 was not initialized!
[火  6月 23 02:16:25 2020] uvcvideo 2-1:1.4: Entity type for entity Camera 9 was not initialized!
[火  6月 23 02:16:32 2020] usb 2-1: usb_suspend_both: status 0
[火  6月 23 02:16:32 2020] usb usb2: usb_suspend_both: status 0
[火  6月 23 02:16:32 2020] tegra-xusb 3530000.xhci: entering ELPG
[火  6月 23 02:16:33 2020] tegra-xusb 3530000.xhci: entering ELPG done
[火  6月 23 02:18:42 2020] tegra-xusb 3530000.xhci: exiting ELPG
[火  6月 23 02:18:42 2020] tegra-xusb 3530000.xhci: Firmware timestamp: 2019-10-31 05:52:15 UTC, Version: 55.17 release
[火  6月 23 02:18:42 2020] tegra-xusb 3530000.xhci: exiting ELPG done
[火  6月 23 02:18:42 2020] usb usb1: usb_suspend_both: status 0
[火  6月 23 02:18:43 2020] usb 2-1: USB disconnect, device number 11
[火  6月 23 02:18:43 2020] usb usb2: usb_suspend_both: status 0
[火  6月 23 02:18:44 2020] tegra-xusb 3530000.xhci: entering ELPG
[火  6月 23 02:18:44 2020] tegra-xusb 3530000.xhci: entering ELPG done
[火  6月 23 02:18:48 2020] tegra-xusb 3530000.xhci: exiting ELPG
[火  6月 23 02:18:48 2020] tegra-xusb 3530000.xhci: Firmware timestamp: 2019-10-31 05:52:15 UTC, Version: 55.17 release
[火  6月 23 02:18:48 2020] tegra-xusb 3530000.xhci: exiting ELPG done
[火  6月 23 02:18:48 2020] usb 2-1: new SuperSpeed USB device number 12 using tegra-xusb
[火  6月 23 02:18:48 2020] usb 2-1: New USB device found, idVendor=8086, idProduct=0b64
[火  6月 23 02:18:48 2020] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[火  6月 23 02:18:48 2020] usb 2-1: Product: Intel(R) RealSense(TM) 515
[火  6月 23 02:18:48 2020] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Camera
[火  6月 23 02:18:48 2020] usb 2-1: SerialNumber: 00000000F0172443
[火  6月 23 02:18:48 2020] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[火  6月 23 02:18:48 2020] uvcvideo: Unknown video format 20202043-0000-0010-8000-00aa00389b71
[火  6月 23 02:18:48 2020] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 515 (8086:0b64)
[火  6月 23 02:18:48 2020] uvcvideo: Unable to create debugfs 2-12 directory.
[火  6月 23 02:18:48 2020] uvcvideo: Unable to create debugfs 2-12 directory.
[火  6月 23 02:18:48 2020] uvcvideo 2-1:1.0: Entity type for entity Extension 3 was not initialized!
[火  6月 23 02:18:48 2020] uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
[火  6月 23 02:18:48 2020] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
[火  6月 23 02:18:48 2020] input: Intel(R) RealSense(TM) 515 as /devices/3530000.xhci/usb2/2-1/2-1:1.0/input/input13
[火  6月 23 02:18:48 2020] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 515 (8086:0b64)
[火  6月 23 02:18:48 2020] uvcvideo: Unable to create debugfs 2-12 directory.
[火  6月 23 02:18:48 2020] uvcvideo 2-1:1.4: Entity type for entity Processing 10 was not initialized!
[火  6月 23 02:18:48 2020] uvcvideo 2-1:1.4: Entity type for entity Extension 11 was not initialized!
[火  6月 23 02:18:48 2020] uvcvideo 2-1:1.4: Entity type for entity Camera 9 was not initialized!
[火  6月 23 02:18:56 2020] usb 2-1: usb_suspend_both: status 0
[火  6月 23 02:18:56 2020] usb usb2: usb_suspend_both: status 0
```

atinfinity, 2020-06-22:
I added `-DFORCE_RSUSB_BACKEND=ON` to CMake flag.

```shell
$ cmake ../ -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=ON
$ make -j4
$ sudo make install
```

As a result, `rs-enumerate-devices` could detect L515.

```
$ rs-enumerate-devices
Device info: 
    Name                          : 	Intel RealSense L515
    Serial Number                 : 	00000000f0172443
    Firmware Version              : 	01.04.01.00
    Recommended Firmware Version  : 	01.04.01.00
    Physical Port                 : 	2-1-12
    Debug Op Code                 : 	15
    Product Id                    : 	0B64
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	L500
    Asic Serial Number            : 	00000003a969845b
    Firmware Update Id            : 	00000003a969845b
```

dorodnic, 2020-06-22:
Hi @atinfinity 
Glad to hear it worked, sorry about the issue. If I recall correctly, Jetson Linux distributions come without hid-iio kernel module, causing cameras with on-board IMU (at the time of writing D435i and L515) to not enumerate. We recommend using `-DFORCE_LIBUVC=ON` flag in Jetson installation guide (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) 
Today you should also be able to install 2.35.2 via apt-get.

atinfinity, 2020-06-23:
@dorodnic Thank you for your information. I understood the background.
So, I closed this issue.

> Today you should also be able to install 2.35.2 via apt-get.

Thanks for the good news! But, I hope to enable CUDA for better performance. 
So, I'll build librealsense from source code.

closed 2020-06-23 by atinfinity