realsenseai/librealsense #6657: Where is rs-convert.exe saving my PNGs?

CarlSeeTwo, 2020-06-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415| 
| Firmware Version                   | 5.12.3.0 | 
| Operating System & Version |  Windows 10  | 
| Kernel Version (Linux Only)    | N/A   | 
| Platform                                 | PC  |
| SDK Version                            |  2.0        | 
| Language                            |  C++                     | 
| Segment			|                    | 

### Issue Description
I am trying to extract some of the depth PNG files from a BAG file using the rs-convert.exe tool.

This is what I provide the tool, and the output at the command line:
[image]

I tried with and without the .PNG extension, I passed a directory, but I don't see any PNG files being made.  I searched my C: drive for the test name (CMDtest), and I don't find anything anywhere.

The BAG file is in the same directory as the rs-convert.exe.

Can anybody help me understand what is going on here?

MartyG-RealSense, 2020-06-20:
Could you try this command please to extract to your CMDtest folder?

rs-convert.exe -d -p C:\Program Files (x86)\Intel RealSense SDK 2.0\tools\CMDtest -i CMD-up-sample.bag

CarlSeeTwo, 2020-06-20:
[image]

I actually just created a folder on my Desktop (so that there are no spaces in the directory), and it does save the PNGs there.

[image]

However, I have encountered a new, completely separate issue (so may require a new thread altogether).

I altered the colorizer in the rs-record-playback to be a "grey-scaled" version like this:

[image]

It looks exactly like what I want in the playback:
[image]

But the PNGs are not the grey-scale that I am looking for:
[image]

Maybe the colorizer is only being applied to the playback, but not what is being saved to the bag file?  And maybe this should be put in a new thread, I am not sure.

MartyG-RealSense, 2020-06-20:
I recall a previous case that stated that rs-convert applies a colorizer to the images.

https://github.com/UnaNancyOwen/rs_bag2image/issues/7#issuecomment-466078090

To get monochome PNGs, a similar bag extraction tool created by a RealSense community member called **rs_bag2image** can be used.

https://github.com/UnaNancyOwen/rs_bag2image

CarlSeeTwo, 2020-06-20:
Sorry to keep being a bother about all of this, but I am still pretty new and I don't have any real experience with CMake.

I have cloned the "rs_bag2image" and I have the folder. With:
[image]

When I try to run CMake on it, I get this error:
[image]

Is there something related to RealSense and CMake that I have not installed?  I cannot find either of the files it calls for in that error.

Very sorry if this is bothersome.  I wish I understood this better so I could ask better questions.

MartyG-RealSense, 2020-06-20:
I have not installed rs_bag2image myself.  It is dependent on the **OpenCV** vision computing software though.  Do you have OpenCv installed on your computer?

CarlSeeTwo, 2020-06-20:
I did not.  I just went to the OpenCV Sourceforge website and followed these instructions:

[image]

[image]

I tried to run the CMake again, but same error.  I am not sure if I even installed the OpenCV properly?  I don't see it as a program in my windows search bar.

MartyG-RealSense, 2020-06-20:
Let's see if there is an easier approach to the image extraction problem.  When you are extracting images from the bag, do you need images of all of the frames or are you trying to just get a PNG of one particular frame?

CarlSeeTwo, 2020-06-20:
I need all or most of the frame images.

I am doing a gesture recognition project for school.  My partners are working on the CNN, and I have been stuck working with the camera, which none of us has experience with.  I've spent weeks trying to get what I need, and when I began I thought that getting the frames and filtering them might be a few days of work.  So it has been a bit frustrating that I don't understand how to use the sample programs appropriately, and I am very far away from being able to build my own programs.

Ideally, I would be able to:
- Capture all or most of the frames in a 1-2 second window (the gesture)
- Filter out the parts of the frames that are not related to the gesture.
- Grey-scale the frames
- Scale them to some specified resolution

Then we would train our CNN with these frames.

For the purpose of this post, I would just like to be able to record a gesture (which I can do to that bag file), and at least get all/most of the frames in PNG format, grey-scaled.

I will need to figure out the filtering, and resolution altering at a later date.

MartyG-RealSense, 2020-06-20:
You and your group may be able to save a lot of time and effort if you use free and open source software called **OpenVINO Toolkit** that is based on CNN and can work with RealSense cameras.

https://software.intel.com/content/www/us/en/develop/tools/openvino-toolkit.html

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openvino

UnaNancyOwen, 2020-06-21:
@CarlSeeTwo Hi, I'm a developer of rs_bag2image.
If you want to use rs_bag2image on Windows, You don't have to build it yourself. You can download pre-built rs_bag2image for Windows from release page. It is able to use right away.
https://github.com/UnaNancyOwen/rs_bag2image/releases/tag/v0.1.3

MartyG-RealSense, 2020-06-21:
@UnaNancyOwen   Thanks so much for your help for @CarlSeeTwo     :)

MartyG-RealSense, 2020-07-05:
Hi @CarlSeeTwo   Do you still require assistance with this case please, or can it be closed?   Thanks!

closed 2020-07-09 by dorodnic

YuvalGofer, 2024-01-31:
> [image]
>
> I actually just created a folder on my Desktop (so that there are no spaces in the directory), and it does save the PNGs there.
>
> [image]
>
> However, I have encountered a new, completely [...]

Hi [CarlSeeTwo], I'm still searching for the converted files. did you copy the bag file in the same directory where rs-convert file is? 
my cmd action look like this: 
rs-convert.exe -i C:\Users\afirnd\Documents\Test1.bag -v C:\Users\afirnd\Desktop\PNGtest\test1 -d
Did I do something wrong?

MartyG-RealSense, 2024-01-31:
Hi @YuvalGofer  I note that you used the -v parameter in your command.  This converts the frames into **.csv** format instead of PNG, and this export mode sometimes does not output files.  To output PNG files, you should use -p instead of -v.

YuvalGofer, 2024-02-04:
Hi Marty,
thanks for your response.
I notice that when I use -p for PNG  it's not output any file's.
any idea why?
Yuval

בברכה,
יובל גופר.
-----------------
0546693207

‫בתאריך יום ד׳, 31 בינו׳ 2024 ב-14:52 מאת ‪MartyG-RealSense‬‏ <‪
***@***.***‬‏>:‬

MartyG-RealSense, 2024-02-05:
@YuvalGofer  Sometimes rs-convert does not work and there is not a clear reason why.  When that happens, if you want to export PNG files then the only action that can be taken is to build from C++ source code an alternative bag extraction tool created by a RealSense user called rs-bag2image.  

https://github.com/UnaNancyOwen/rs_bag2image

This program exports depth frames as monochrome PNG images instead of the colorized ones that rs-convert outputs.  If you need colorized images then the program's creator provides advice about modifying the code for colorization at https://github.com/UnaNancyOwen/rs_bag2image/issues/7#issuecomment-466399057

YuvalGofer, 2024-02-12:
> colorization at UnaNancyOwen/rs_bag2image#7 (comment)

Thanks for the tip!

בברכה,
יובל גופר.
-----------------
0546693207

‫בתאריך יום ב׳, 5 בפבר׳ 2024 ב-10:31 מאת ‪MartyG-RealSense‬‏ <‪
***@***.***‬‏>:‬