realsenseai/librealsense #6658: How to patch for custom kernel?

zonetrooper32, 2020-06-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Due to drivers and performance issue, I am forced to use custom linux kernel with patches (more specifically, linux-surface kernel 5.3, address: https://github.com/linux-surface/linux-surface) for my Surface Book 2 with Ubuntu 18.04.04 install. 

However, I noticed that librealsense2-dkms does not support kernel version 5.3.18. I am thinking to compile a custom linux kernel with both surfacelinux patch and the librealsense2 patch applied. The question is: What is the specific linux kernel 5.3 version that librealsense2 supports? And how do I apply librealsense2 kernel patch over my custom kernel compiling?

Thank you!

MartyG-RealSense, 2020-06-20:
Hi @zonetrooper32   the list of officially supported kernels is listed at the top of each Librealsense SDK version's details on the 'Releases' page.

https://github.com/IntelRealSense/librealsense/releases/

[image]

There is the option to build Librealsense source code with a "backend" method that is not dependent on Linux versions or kernel versions and does not require patching.  do you think that this might suit your situation, please?

zonetrooper32, 2020-06-20:
> Hi @zonetrooper32 The list of officially supported kernels is listed at the top of each Librealsense SDK version's details on the 'Releases' page.
> 
> https://github.com/IntelRealSense/librealsense/releases/
> 
> [image]
> 
> There is the option to build Librealsense source code with a "backend" method over an internet connection that is not dependent on Linux versions or kernel versions and does not require patching. Do you think that this might suit your situation, please?

Thank you for the reply. I will search around the "backend" method. In the mean time I'll close the issue.

closed 2020-06-20 by zonetrooper32

MartyG-RealSense, 2020-06-20:
I was asking about backend first to find out whether you were only limited to DKMS package installation.   I'm happy to point you to the backend method's installation details.  :)

After downloading a zip file of an SDK version's source code from the "Assets" list at the bottom of each SDK version's details on the 'Releases' web-page and extracting the zip's contents, you can install with the backend method using the instructions in the link below.

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

zonetrooper32, 2020-06-21:
> I was asking about backend first to find out whether you were only limited to DKMS package installation. I'm happy to point you to the backend method's installation details. :)
> 
> After downloading a zip file of an SDK version's source code from the "Assets" list at the bottom of each SDK version's details on the 'Releases' web-page ( https://github.com/IntelRealSense/librealsense/releases/ ) and extracting the zip's contents, you can install with the backend method using the instructions in the link below.
> 
> [#6368 (comment)](https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279)
> 
> Good luck!

Thank you again for the link. I have successfully compiled following the instructions.

However, I do have one extra question, if using backend installation, will it influence the frame metadata? From what I read the purpose of patching around is to make sure that the retrieved frame metadata is accurate.

MartyG-RealSense, 2020-06-21:
Great news - thanks for the update!

The backend method includes support for metadata, which would otherwise have to be patched in.

The link below describes the advantages and disadvantages of the backend installation method compared to the usual methods such as compiling from source.  Please scroll down the linked-to comment to the section headed '**What are the advantages and disadvantages of using libuvc vs patched kernel modules?**'.

https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604

zonetrooper32, 2020-06-21:
> Great news - thanks for the update!
> 
> The backend method includes support for metadata, which would otherwise have to be patched in. I am not aware of negative consequences compared to patched-in metadata support.
> 
> The link below describes the advantages and disadvantages of the backend installation method compared to the usual methods such as compiling from source. Please scroll down the linked-to comment to the section headed '**What are the advantages and disadvantages of using libuvc vs patched kernel modules?**'.
> 
> [#5212 (comment)](https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604)

Thank you, those are indeed useful information.