realsenseai/librealsense #6659: SDK UWP support status

wdelcloy, 2020-06-21:
| Required Info                         |                                                                    |
|-------------------------------|-------------------------------------------   |
| Camera Model                       |                                                                    | 
| Firmware Version                   |                                                                    | 
| Operating System & Version  |  Win 10                                                       | 
| Kernel Version (Linux Only)    |                                                                    | 
| Platform                                | PC                                                               |
| SDK Version                          |  2.35.2                                                         | 
| Language                             |  C#                                                               | 
| Segment			      |  AR                                                               | 

### Issue Description
I would like to get a status about the current support of UWP platform.

**D400 family**
By looking at other issues, I have understood that a driver exists for D400 camera family
https://downloadcenter.intel.com/download/28723/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10
So I understand that a developer **can use MediaCapture UWP API to access to D400 camera**, there is a post claiming a successful usage : 
https://github.com/IntelRealSense/librealsense/issues/2702#issuecomment-444035850

**Other families**
Today, **I understand that there is no support at all for other families than D400**
In particular, for T265, support says "There is not a UWP (DMFT) release for T265" 
https://github.com/IntelRealSense/librealsense/issues/3758#issue-433225059
https://github.com/IntelRealSense/librealsense/issues/5774#issue-559333380

**Forecast**
I understand that usage of native realsense2.dll together with wrapper Intel.Realsense.dll is today only possible for Windows Desktop applications, not for UWP applications. 
Is there a plan to provide realsense SDK compatible with UWP platform ?
Or to provide an access to all cameras through UWP MediaCapture API ? (I'm particularly interested in T265)

MartyG-RealSense, 2020-06-21:
The UWP driver for the 400 Series cameras does receive occasional updates (the most recent being April 2020 at the time of writing this).  UWP support is useful for using RealSense with UWP-supporting products such as Microsoft Hololens.  There does not seem to be widespread use of it amongst RealSense users though.

I am on the Support team and not the developer team and so do not have knowledge of plans for UWP.  My expectation would be that support for it is unlikely to be expanded beyond the 400 Series though.

MartyG-RealSense, 2020-07-05:
Case closed due to no futher comments received and no further information available about the status of UWP support.

closed 2020-07-05 by MartyG-RealSense