realsenseai/librealsense #6662: set_region_of_interest get a error

LW-SCU, 2020-06-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.03.00| 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-96-generic      | 
| Platform                                 | PC  |
| SDK Version                            |    v    2.33.1                  | 
| Language                            |  C++            | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I get and set ROI for color sensor by https://github.com/IntelRealSense/librealsense/issues/4363#issuecomment-508405760
This is my code:
```c++
pipeline pipe;
config cfg;
cfg.enable_stream(RS2_STREAM_COLOR,1280,720,RS2_FORMAT_BGR8,30);
cfg.enable_stream(RS2_STREAM_DEPTH,480,270,RS2_FORMAT_Z16,30);
rs2::pipeline_profile profile = pipe.start(cfg);
roi_sensor rgb_roi_sensor(profile.get_device().first<color_sensor>());
region_of_interest roi_interest = rgb_roi_sensor.get_region_of_interest();
printf("%d,%d,%d,%d\n",roi_interest.min_x,roi_interest.min_y,roi_interest.max_x,roi_interest.max_y);
rgb_roi_sensor.set_region_of_interest(roi_interest);
```
I get a ROI and set it ,but i get a error, how do i go to set ROI correctly?
```c
320,180,960,540
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x75 failed. Error type: Invalid parameter (-6).
```

MartyG-RealSense, 2020-06-22:
There is a previous case of this error occurring when getting / setting ROI.  In that case, advice for checking whether performing this action is supported by your camera is provided in the link below:

https://github.com/IntelRealSense/librealsense/issues/5308#issuecomment-560132907

LW-SCU, 2020-06-22:
@MartyG-RealSense hello,
my camera is D435i, it has RGB sensor.

My new code is:
```c++
int main(){
    pipeline pipe;
    config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR,1280,720,RS2_FORMAT_BGR8,30);
    cfg.enable_stream(RS2_STREAM_DEPTH,480,270,RS2_FORMAT_Z16,30);
    auto profile = pipe.start(cfg);

    device _device = profile.get_device();
    depth_sensor _depth_sensor = _device.first<depth_sensor>();
    if(_depth_sensor.is<roi_sensor>()){
        cout<<"support depth sensor roi"<<endl;
        region_of_interest _region({60,33,420,237});
        roi_sensor _roi_sensor = _depth_sensor.as<roi_sensor>();

        auto _pre_region = _roi_sensor.get_region_of_interest();
        cout<<"depth sensor roi:"<<_pre_region.min_x<<"\t"<<_pre_region.min_y
            <<"\t"<<_pre_region.max_x<<"\t"<<_pre_region.max_y<<endl;

        _roi_sensor.set_region_of_interest(_region);

        _pre_region = _roi_sensor.get_region_of_interest();
        cout<<"depth sensor roi:"<<_pre_region.min_x<<"\t"<<_pre_region.min_y
            <<"\t"<<_pre_region.max_x<<"\t"<<_pre_region.max_y<<endl;

        cout<<"set depth sensor roi successfully"<<endl;
    }
    cout<<endl;
    color_sensor _color_sensor = _device.first<color_sensor>();
    if(_color_sensor.is<roi_sensor>()){
        cout<<"support color sensor roi"<<endl;
        region_of_interest _region({320,180,960,540});
        roi_sensor _roi_sensor = _color_sensor.as<roi_sensor>();

        auto _pre_region = _roi_sensor.get_region_of_interest();
        cout<<"color sensor roi:"<<_pre_region.min_x<<"\t"<<_pre_region.min_y
            <<"\t"<<_pre_region.max_x<<"\t"<<_pre_region.max_y<<endl;

        _roi_sensor.set_region_of_interest(_region);
        cout<<"set color sensor roi successfully"<<endl;

        _pre_region = _roi_sensor.get_region_of_interest();
        cout<<"color sensor roi:"<<_pre_region.min_x<<"\t"<<_pre_region.min_y
            <<"\t"<<_pre_region.max_x<<"\t"<<_pre_region.max_y<<endl;
    }
    pipe.stop();
}
```
This code set ROI successfully for depth sensor, but failed for color sensor. And I got  a same error.
```c
support depth sensor roi
depth sensor roi:60	33	420	237
depth sensor roi:60	33	420	237
set depth sensor roi successfully

support color sensor roi
color sensor roi:320	180	960	540
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x75 failed. Error type: Invalid parameter (-6).
```

MartyG-RealSense, 2020-06-22:
Are you able to stream an RGB color image through a program such as the RealSense Viewer, please?

ev-mp, 2020-06-22:
@LW-SCU , can you also verify that the Auto-Exposure control for RGB sensor is turned on ?

closed 2020-06-30 by LW-SCU