realsenseai/librealsense #6665: Project point coordinate to pixel coordinate using MATLAB

mk-kor, 2020-06-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model | {D415} |
| Firmware Version | 05.12.03.00 |
| Operating System & Version | Win 10 |
| Kernel Version (Linux Only) | (e.g. 4.14.13) |
| Platform | PC |
| SDK Version | 2.33.1 |
| Language | Matlab} |
| Segment | Robot |

Issue Description

Hi !

I'm trying to project Target's 3D point to Pixel coordinate.
I believe there is method for projection , but I cant find any related.
May u let me know any documents or something like that?

I know how to project with intrinsic parameters of camera and I found some of about Intrinsic Params though. Thing is, I'm not familiar with c++ codes...... and Its pretty hard to understand wrappers.
Is there any example codes?

My questions are.
1. Is there any method for projection using MATLAB?
2. Is there any sample code to get Intrinsics using MATLAB?

Thanks in advance !

MartyG-RealSense, 2020-06-22:
1.  MATLAB has a built-in **worldToImage** function for projecting 3D world points to a 2D image.

https://uk.mathworks.com/help/vision/ref/worldtoimage.html  

2.  I hope the link below will be useful in getting intrinsics information using the MATLAB wrapper.

https://github.com/IntelRealSense/librealsense/issues/3735

3.  If by 'perpendicular to target' depth you mean measuring the distance between two points, built-in functions in MATLAB may again be able to help with this.

https://uk.mathworks.com/matlabcentral/answers/16848-how-to-find-distance-between-two-points

mk-kor, 2020-06-24:
I got u. Thanks a lot !

closed 2020-06-24 by mk-kor