realsenseai/librealsense #6667: can I change exposure mode by realsense viewer or lanuch file

Dunkem, 2020-06-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400I | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version | Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0                                         | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  legacy / 2.<?>.<?>                         | 
| Language                            |  c++                           | 
| Segment			|  Robot                           | 

### Issue Description
I want to set the infra and color exposure（and I set it in realsense viewer），but it looks like not work. So can you tell me how to set the exposure value by realsense viewer or by launch file(I didn't find any info about exposure in rs_camera.launch)

MartyG-RealSense, 2020-06-22:
Hi @Dunkem   In the RealSense Viewer, you can find manual exposure controls for the depth and infrared streams by expanding open the **Controls** section of the **Stereo Module** section of the Viewer's options side-panel.  Changing the exposure value will automatically turn off auto-exposure.  

You can re-enable the Stereo Module's auto-exposure at any time by left-clicking on the box beside the **Enable Auto Exposure** option to turn the box from black (off) to blue (on).

[image]

You can likewise find the color exposure controls by expanding open the **Controls** section of the **RGB Camera** settings in the side-panel.

[image]

Once the streams have been enabled then making changes to the exposure settings will update the exposure of images in the center panel of the Viewer in real-time.

[image]

In ROS, exposure can be changed in real-time whilst the stream is active by using a feature called **dynamic reconfigure**, which is also known as **rqt-reconfigure**.

https://github.com/IntelRealSense/realsense-ros#set-camera-controls-using-dynamic-reconfigure-params

Dunkem, 2020-06-22:
thanks for your reply, but I set in realsense viewer and then close realsense viewer, run roslaunch realsense_camera rs_camera.launch, it looks like the config about exposure doesn't work. As what you reply I guess when I change the config info in realsense viewer but it doesn't write in camera. So for my application the only way for me to change the exposure mode is rqt_reconfigure?

MartyG-RealSense, 2020-06-22:
The RealSense Viewer and the ROS wrapper are separate systems.  Changes that you make to one will not affect the other.  So if you are using ROS, you should change exposure during runtime with rqt-reconfigure.

Dunkem, 2020-06-22:
ok, I misunderstand it. But I run ROS wapper by ssh which does contain a GUI desktop, it means I can't use rqt_reconfure. So if there exists another way for me to change the exposure mode and exposure value.
thanks

MartyG-RealSense, 2020-06-22:
Setting the exposure manually in ROS with a 400 Series camera before streaming seems to be more complicated than doing it with rqt_reconfigure.  Apparently it can be done with a Python node script though.

https://github.com/IntelRealSense/realsense-ros/issues/345

MartyG-RealSense, 2020-06-22:
You can use C++ node scripts with RealSense ROS too, yes.  I will see if I can find a C++ equivalent to the above instruction.

Dunkem, 2020-06-22:
ok, thank you for your reply

MartyG-RealSense, 2020-06-22:
Having looked at the subject, I believe that creating a C++ ROS script to change exposure may be so hard that it might be far easier to set the exposure with a camera configuration file called a **json** that can be called from the launch file.

https://github.com/IntelRealSense/realsense-ros/issues/411#issuecomment-477463559

You can create a custom json file easily by configuring values in the RealSense Viewer and then using a json export option on the toolbar on top of the options side-panel.

Dunkem, 2020-06-22:
[image]
I do what you told me, but I get error when I run roslaunch realsense2_camera rs_camera.launch

MartyG-RealSense, 2020-06-22:
Please tell me what you did to get this error.

Dunkem, 2020-06-22:
1. open realsense viewer and set the exposure value in stereo_module
2. save this config as a json file by what you told me.
3. set rs_camera.launch <arg name="json_file_path"      default="path where I save the json file"/>

Dunkem, 2020-06-22:
{
    "aux-param-autoexposure-setpoint": "1536",
    "aux-param-colorcorrection1": "0.298828",
    "aux-param-colorcorrection10": "-0",
    "aux-param-colorcorrection11": "-0",
    "aux-param-colorcorrection12": "-0",
    "aux-param-colorcorrection2": "0.293945",
    "aux-param-colorcorrection3": "0.293945",
    "aux-param-colorcorrection4": "0.114258",
    "aux-param-colorcorrection5": "-0",
    "aux-param-colorcorrection6": "-0",
    "aux-param-colorcorrection7": "-0",
    "aux-param-colorcorrection8": "-0",
    "aux-param-colorcorrection9": "-0",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "8500",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "166",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-laserpower": "0",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "24",
    "param-maxscorethreshb": "2047",
    "param-medianthreshold": "500",
    "param-minscorethresha": "1",
    "param-neighborthresh": "7",
    "param-raumine": "1",
    "param-rauminn": "1",
    "param-rauminnssum": "3",
    "param-raumins": "1",
    "param-rauminw": "1",
    "param-rauminwesum": "3",
    "param-regioncolorthresholdb": "0.0499022",
    "param-regioncolorthresholdg": "0.0499022",
    "param-regioncolorthresholdr": "0.0499022",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "10",
    "param-robbinsmonroincrement": "10",
    "param-rsmdiffthreshold": "4",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.375",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "60",
    "param-scanlinep1onediscon": "105",
    "param-scanlinep1twodiscon": "70",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "190",
    "param-scanlinep2twodiscon": "130",
    "param-secondpeakdelta": "325",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-ir-format": "R8L8",
    "stream-width": "1280"
}
this is the json file which I save from realsense viewer

Dunkem, 2020-06-22:
I debug it and find it throw this error at **adv.load_json(json_file_content);**

MartyG-RealSense, 2020-06-22:
You can set the json path by adding it onto the end of the launch instruction instead of putting the path in the json file itself.   For example:

**roslaunch realsense2_camera rs_camera.launch json_file_path:=/path/to/config/DefaultPreset_viewer.json`**

Dunkem, 2020-06-22:
I do this but I get the same error

MartyG-RealSense, 2020-06-22:
Research of the **set_pu(id=4) failed** error suggests that it may occur because of a problem in the json file, such as a problem with loading it or a corruption.  What was the **json_file_path** command that you used in the launch instruction, please?

Dunkem, 2020-06-23:
**realsense2_camera rs_camera.launch json_file_path:=/path/test.json**
this is the way I load json file, and the detail info about this file is put above!

MartyG-RealSense, 2020-06-23:
I have been over the information carefully but cannot see where the problem is occurring.  I will ask the ROS wrapper developer.

@doronhi  Can you see why @Dunkem might be having difficulty calling on the json file from the launch statement above, please?   Thanks!

MartyG-RealSense, 2020-07-05:
Hi @Dunkem   Do you still require assistance with this case, please?

Dunkem, 2020-07-05:
I find another method that meets my needs， but it isn't a long-term solution. If there have an alternate will be better

MartyG-RealSense, 2020-07-05:
Re-reading the whole case, I note that the reason that you could not use rqt_reconfigure to change exposure was that you did not have a display due to using SSH.  Would it be possible to set the exposure with a **rosparam** instruction in the command prompt?  Like the one below:

**rosparam set /camera/realsense_ros_camera_manager/rs435_color_exposure 1000**

Dunkem, 2020-07-05:
now I use this method, but this maybe not the best way to solve my case. Because every time I run my project, I will type this command in my terminal.

MartyG-RealSense, 2020-07-05:
I did some further research and found information for setting it with a C++ node or the launch file.

**C++**
void:ros::param::set(parameter_name, input_value)

**From launch file**
<param name="param-name" value="param-value" />

http://www.clearpathrobotics.com/assets/guides/kinetic/ros/ROS%20Parameter%20Server.html

Dunkem, 2020-07-05:
in my case, I use rqt_reconfigure through multi-master. So it's ok for display

Dunkem, 2020-07-05:
OK, I will try this way, thank you for your help

MartyG-RealSense, 2020-07-05:
You are very welcome.  Let us know how your tests go, please   :)

Dunkem, 2020-07-08:
it can work in my project, thanks for your help!!!

closed 2020-07-08 by Dunkem

MartyG-RealSense, 2020-07-08:
Awesome news - thanks so much for the update about your success   :)