realsenseai/librealsense #6669: How to display the depth based on converted csv or png depth files

EvanZouQ, 2020-06-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |up-tp-date | 
| Operating System & Version | win10  |                                   | 
| Platform                                 | PC |
| SDK Version                            |2.33.1 }                          | 
| Language                            | python }                          | 
| Segment			|  object detection }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello everyone! Since I get issues on direct loading of bag file, I use rs-covert tool to tranfer bag file into csv and png files which contain depth information. So how can I display the depth data using python? Thanks.

MartyG-RealSense, 2020-06-22:
Hi @EvanZouQ   Have you tried using the **read_bag_file** Python example program to read the depth information from the bag file, please?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py

EvanZouQ, 2020-06-23:
Dear MartyG, I have performed this example using Pycharm. It returns 'No input parameter have been given.
For help type --help'. How can I deal with this? Thanks

MartyG-RealSense, 2020-06-23:
I researched the **read_bag_example.py** program.  It looks for an '-i' input reference and then the bag file name.  

For example, if the bag file is in the same directory as the read_bag_example.py program and the name of your bag file was **test.bag,** this command would be used to run it:

read_bag_example.py -i test.bag

EvanZouQ, 2020-06-23:
Thanks for your reply. However, after manipulation, it seems that there is no images. When should I do after this operation? Thanks again.

MartyG-RealSense, 2020-06-23:
Do you mean that when the bag file plays, no images are shown on the screen please?

EvanZouQ, 2020-06-23:
Yeah.  After I performed this command, there is no images on the screen. Thanks.

MartyG-RealSense, 2020-06-23:
I have read that **read-bag-example.py** has been successfully tested with the **outdoors.bag** sample bag file.  This is the first of the files on the RealSense sample bag page, described as "Outdoors scene captured with D415 pre-production sample (Depth from Stereo)"

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

EvanZouQ, 2020-06-23:
Thanks for your reply. I mean that I want get depth data and display these information on the screen using pre-recorded bag or csv file. Thanks.

MartyG-RealSense, 2020-06-23:
**read_bag_example.py** may not be the best approach if you want to display the distance on screen.  A more suitable tutorial may be Intel's **distance_to_object** Pyrealsense2 tutorial, which aligns RGB and depth frames read from a bag file and displays a distance measurement on-screen.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

EvanZouQ, 2020-06-23:
Thanks for this links.  It can be successfully performed by pycharm. Thanks.

MartyG-RealSense, 2020-07-05:
Case closed due to solution being found.

closed 2020-07-05 by MartyG-RealSense