realsenseai/librealsense #6673: Point cloud weird texture mapping

hojunson, 2020-06-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I am wondering why the pointcloud example in the pyhton wrapper looks like this(mirrored?)

[image]

In the realsense view, it looks okay. 
[image]

ev-mp, 2020-06-23:
@hojunson , I'm sorry but I cannot see any mirroring in the snapshots - things in and next to the closet appear in the same order.

MartyG-RealSense, 2020-06-23:
Hi @ev-mp  The main problem I observe in the image provided by @hojunson is that on the Python wrapper version of the image, the bottom of the image ends up on top of the wardrobe (highlighted in green below).

[image]

MartyG-RealSense, 2020-07-05:
Hi @hojunson   Do you still require assistance with this case, please?   Thanks!

hojunson, 2020-07-07:
No. I think it was a sort of visualization problem.

MartyG-RealSense, 2020-07-07:
@hojunson   Okay, thank you very much for the update.  Please click the 'Close and comment' button beneath the comment writing box if you are happy to close the case.  Thanks again!

closed 2020-07-18 by MartyG-RealSense