realsenseai/librealsense #6674: Tracking position with imu data

vams08, 2020-06-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<I am working with imu data given by this device i am trying to find position of camera placed on moving object after certain amount of time and i want it to be accurate in centimeter scale is this possible or any one worked with similar prblm you can say your approach, I want to estimate postion only with help of acceleration data and i tried few methods like double integration
 >

MartyG-RealSense, 2020-06-23:
Hi @vams08   Which camera model are you using, please?

If it is a D435i and you want to get the current position of the camera on a moving object, then obtaining it through **localization** with SLAM may be an option.  Intel have published a guide to SLAM with D435i and ROS.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

MartyG-RealSense, 2020-07-05:
Hi @vams08   Do you still require assistance with this case, please?

closed 2020-07-06 by vams08