realsenseai/librealsense #6675: Add BUILD_WRAPPERS option

oiramario, 2020-06-23:
[add-wrappers-option.patch.txt](https://github.com/IntelRealSense/librealsense/files/4817758/add-wrappers-option.patch.txt)

What's your point?

MartyG-RealSense, 2020-06-23:
Hi @oiramario   Could you tell us more about your request, please?

Is the **add-wrappers-option.patch.txt** file code that you would like added to the RealSense SDK to improve its wrapper building support?

If it is, suggestions for new code to be merged into the SDK are submitted on the **Pull Request** page:

https://github.com/IntelRealSense/librealsense/pulls

Thanks!

oiramario, 2020-06-24:
I've do that.
https://github.com/IntelRealSense/librealsense/pull/6687

MartyG-RealSense, 2020-06-24:
Thanks so much for your contribution to the SDK!

ev-mp, 2020-06-29:
@oiramario , following-up on the mentioned PR status we'll close this accordingly.

closed 2020-06-29 by ev-mp