realsenseai/librealsense #6677: how can i use export_to_ply functions to save ply as ascii format?

wuhaoqi-hub, 2020-06-23:
I want to export ply as ascii format,but i can't find some methods to slove this problem,so i want to consult this question,this is my code.
int main()
{
	context ctx;
	int ind(0);

	for (auto&& dev : ctx.query_devices())
	{
		pipeline pipe(ctx);
		config cfg;
		cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
		cfg.enable_stream(RS2_STREAM_ANY, 0, 640, 480, RS2_FORMAT_Z16, 30);
		cfg.enable_stream(RS2_STREAM_ANY, 1, 640, 480, RS2_FORMAT_Y8, 30);
		cfg.enable_stream(RS2_STREAM_ANY, 2, 640, 480, RS2_FORMAT_Y8, 30);
		pipeline_profile selection = pipe.start(cfg);
		device selected_device = selection.get_device();

		advanced_mode advanced_device(selected_device);

		depth_sensor depth_sensor = selected_device.first<rs2::depth_sensor>();
		//	option_range range = depth_sensor.get_option_range(RS2_OPTION_DEPTH_UNITS);
		depth_sensor.set_option(RS2_OPTION_DEPTH_UNITS, 0.001);

		STDepthTableControl depth_table = advanced_device.get_depth_table();
		depth_table.depthClampMax = 1000;
		depth_table.depthUnits = 1000;
		advanced_device.set_depth_table(depth_table);

		pointcloud pc;
		points points;

		frameset frames = pipe.wait_for_frames();
		video_frame infrared = frames.get_color_frame();
		if (!infrared)
			infrared = frames.get_infrared_frame();
		depth_frame depth = frames.get_depth_frame();

		pc.map_to(infrared);
		points = pc.calculate(depth);

		char path[200];
		sprintf_s(path, "F:\\20191024_%d.ply", ind);
		points.export_to_ply(path, infrared);

		++ind;
	}
	return EXIT_SUCCESS;
}

dorodnic, 2020-06-23:
Hi @wuhaoqi-hub 
This is not possible with `rs2::points::export_to_ply` method but you can use `save_to_ply` class from [rs_export.hpp](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_export.hpp) - 
```cpp
rs2::save_to_ply exporter(path, pc);
exporter.set_option(rs2::save_to_ply::OPTION_PLY_BINARY, 0.f);
exporter.export_to_ply(points, infrared);
```

wuhaoqi-hub, 2020-06-23:
thank you very much

wuhaoqi-hub, 2020-06-26:
i use the method you answered  for me,but the  function  "rs2::save_to_ply::export_to_ply" can't be visited.

MartyG-RealSense, 2020-06-26:
@wuhaoqi-hub   I note that you included **export_to_ply** in your instruction.  @dorodnic stated above that using export_to_ply was not possible. and suggested this code instead:

`rs2::save_to_ply exporter(path, pc);
exporter.set_option(rs2::save_to_ply::OPTION_PLY_BINARY, 0.f);
exporter.export_to_ply(points, infrared);`

wuhaoqi-hub, 2020-06-27:
I used the code exporter.export_to_ply(points, infrared);it has export_to_ply,but the export_to_ply is the private function ,the exporter object can't visit it.can you give me some example code？thanks

MartyG-RealSense, 2020-06-27:
I could not find an example for saving ASCII to ply.  @dorodnic will be able to provide better advice about custom code that you can try in your project.

closed 2020-06-27 by wuhaoqi-hub

ilyak93, 2022-04-18:
> @wuhaoqi-hub I could not find an example for saving ASCII to ply. @dorodnic will be able to provide better advice than I can about custom code that you can try in your project.

The function work if you change its `private` to `public`, but I don't get the color mapped correctly to the points, just everything gray.

MartyG-RealSense, 2022-04-18:
Hi @ilyak93  Exporting color to ply usually does not work on Python.  The subject has been researched by me extensively over the years and the only script confirmed to work is one at https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293 that exports color to ply but without vertex normals.

ilyak93, 2022-04-18:
@MartyG-RealSense, thanks, I'm trying it on C++ both methods (using points and using the function proposed here) with no success. I'm using 2.34.0-0~realsense0.2250 SDK version. Any suggestions ?

MartyG-RealSense, 2022-04-18:
@ilyak93  I will research your question tomorrow.  Thanks again for your patience!

ilyak93, 2022-04-18:
@MartyG-RealSense, ok thanks. Should I open a new issue or can we continue here ? 

I have found (by using the function that the exporter exploits)  that:
[image]
that `texcoords` are all set to 0 (the `cout << "here"` loop is my code injected into real sense `rs_export.hpp`, which why probably I get only one color (the color of (u=0,v=0)), which means my pointcloud somehow doesn't know what is his texture coordinates. Maybe it is somehow connected to the fact it is software-device, some data maybe lost, I think the implementation of this function `rs2::points points = pc.calculate(depth_frame2);` (calculate) should give the answers if you can look around it.

Thank you in advance.

MartyG-RealSense, 2022-04-18:
Given that this case was originally about exporting ASCII characters to ply instead of color, please start a new issue and link back to this one.