realsenseai/librealsense #6682: Hi, is there any way to get the physical size of a pixel? That is the height and width of one pixel on the CCD

JimXu1989, 2020-06-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<30>.<0> }                          | 
| Language                            |  {python}                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I want get the line in real world camera coordinate system which corrspond to one pixel in the image, so I need the size of one pixel on the CCD sensor, while, I can not get it from "intrinsics = stream2.get_intrinsics()"

agrunnet, 2020-06-24:
It’s in the spec sheet. 3um x 3um for OV9281 sensor.

JimXu1989, 2020-06-24:
> It’s in the spec sheet. 3um x 3um for OV9281 sensor.

Thanks!!

JimXu1989, 2020-06-24:
> It’s in the spec sheet. 3um x 3um for OV9281 sensor.

Is D435 i use OV9281 sensor?

agrunnet, 2020-06-24:
Yes. And D455 will have same size pixels.

closed 2020-06-24 by agrunnet

JimXu1989, 2020-06-24:
> Yes. And D455 will have same size pixels.

Thanks a lot !!

Steven-ZY, 2020-06-24:
@agrunnet Thanks a lot. But where is the spec sheet？ Thanks in advance.

Steven-ZY, 2020-06-24:
@JimXu1989 Excuse me. I have the same question as you. But if the size of a pixel is 3um*3um, fx=613 pixel, then f=613*3um. It's too small, isn't it?

JimXu1989, 2020-06-24:
> @JimXu1989 Excuse me. I have the same question as you. But if the size of a pixel is 3um x 3um, fx=613 pixel, then f=613 x 3um. It's too small, isn't it?

https://www.mouser.com/datasheet/2/612/Intel-RealSense-D400-Series-Datasheet-1383890.pdf

Steven-ZY, 2020-06-24:
@JimXu1989 Thanks for your replying!

JimXu1989, 2020-06-24:
> @JimXu1989 Thanks for your replying!

哈哈，不客气：）