realsenseai/librealsense #6683: the SDK2.0 v2.35 can't work normally in my computer?

wuhaoqi-hub, 2020-06-24:
I want to use save_to_ply function in my code ,but v2.30 sdk don't has this function ,so i update the SDk to 2.35,but the code which run normally can't run in the SDK v2.35,throw wrong_api_call_sequence_error(e); how can i solve this problem?
my code:
#include <librealsense2/rs.hpp>
#include <librealsense2/rs_advanced_mode.hpp>
#include <iostream>

using namespace rs2;
using namespace rs400;
using namespace std;

int main()
{
	context ctx;
	int ind(0);

	for (auto&& dev : ctx.query_devices())
	{
		pipeline pipe(ctx);
		config cfg;
		cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
		cfg.enable_stream(RS2_STREAM_ANY, 0, 640, 480, RS2_FORMAT_Z16, 30);
		cfg.enable_stream(RS2_STREAM_ANY, 1, 640, 480, RS2_FORMAT_Y8, 30);
		cfg.enable_stream(RS2_STREAM_ANY, 2, 640, 480, RS2_FORMAT_Y8, 30);
		pipeline_profile selection = pipe.start(cfg);
		device selected_device = selection.get_device();

		advanced_mode advanced_device(selected_device);

		depth_sensor depth_sensor = selected_device.first<rs2::depth_sensor>();
		//	option_range range = depth_sensor.get_option_range(RS2_OPTION_DEPTH_UNITS);
		depth_sensor.set_option(RS2_OPTION_DEPTH_UNITS, 0.001);

		STDepthTableControl depth_table = advanced_device.get_depth_table();
		depth_table.depthClampMax = 1000;
		depth_table.depthUnits = 1000;
		advanced_device.set_depth_table(depth_table);

		pointcloud pc;
		points points;

		frameset frames = pipe.wait_for_frames();
		video_frame infrared = frames.get_color_frame();
		if (!infrared)
			infrared = frames.get_infrared_frame();
		depth_frame depth = frames.get_depth_frame();

		pc.map_to(infrared);
		points = pc.calculate(depth);

		char path[200];
		sprintf_s(path, "F:\\20191024_%d.ply", ind);
		points.export_to_ply(path, infrared);

		++ind;
	}
	return EXIT_SUCCESS;
}

MartyG-RealSense, 2020-06-24:
Hi @wuhaoqi-hub  The first action that I would recommend taking is to change your cry stream definitions from "ANY" to the ones below:

cfg.enable_stream(RS2_STREAM_DEPTH, 0, 640, 480, RS2_FORMAT_Z16, 30);
cfg.enable_stream(RS2_STREAM_INFRARED, 1, 640, 480, RS2_FORMAT_Y8, 30);
cfg.enable_stream(RS2_STREAM_INFRARED, 2, 640, 480, RS2_FORMAT_Y8, 30);

wuhaoqi-hub, 2020-06-27:
I solve this problem according to put the DLL in the right place.

closed 2020-06-27 by wuhaoqi-hub

MartyG-RealSense, 2020-06-27:
Great news that you found a solution - thanks for the update!