realsenseai/librealsense #6684: RuntimeError: Couldn't resolve requests error on L515

malith1992, 2020-06-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | 01.04.01.00 | 
| Operating System & Version |   {Ubuntu 18.04  |                                   | 
| Platform                                 |  NVIDIA Jetson Nano.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot}                          | 

### Issue Description

I have tried to run the opencv_viewer_example.py using the realsense L515 camera. But I get the following error.

> Traceback (most recent call last):
  File "opencv_viewer_example.py", line 19, in <module>
    pipeline.start(config)
RuntimeError: Couldn't resolve requests

Please tell me how to fix this issue.

Tachiuo00, 2020-06-26:
I think you need to specify the resolution supported by L515.
For examples, if you rewrite the code as follows, it fix this issue.
[Before]
```
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
```

[After]
```
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
```

malith1992, 2020-06-30:
> I think you need to specify the resolution supported by L515.
> For examples, if you rewrite the code as follows, it fix this issue.
> [Before]
>
> ```
> config.enable_stream(rs.stream.depth, 640, 480, [...]

Can't I get the depth frame and the color frame with the same size ?

Tachiuo00, 2020-07-02:
Please add an align object and apply it.
In opencv_viewer_example.py
`
# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)

# Align objects
align_to = rs.stream.color
align = rs.align(align_to)

# Start streaming
pipeline.start(config)

try:
    while True:
        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        aligned_frames = align.process(frames)
        depth_frame = aligned_frames.get_depth_frame()
        color_frame = aligned_frames.get_color_frame()
`

RealSenseSupport, 2020-07-11:
Hi @malith1992 

Were you able to try @Tachiuo00 's suggestions?

And yes....you should be able to run 640x480 on both depth and color.

Thanks

RealSenseSupport, 2020-07-13:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-07-21 by RealSenseSupport

markliou, 2021-01-28:
I don't know why I meet the same issue.
The solution below did not work. I am not sure if the depth function is abnormal.
But

`
config.enable_stream(rs.stream.depth, 320, 240, rs.format.z16, 30)
`

works for me.