realsenseai/librealsense #6685: What is changed by the tare calibration of D435

LyJoan, 2020-06-24:
Hi.
I know there are some calibrations,including tare calibration,on-chip calibration,dynamic calibration and so on.I have a D435 which  is unacceptable in get_distance.Therefore,I want to calibrate it to solve this problem.For example,I measured a box which distance is 2.0 meters，however the D435 told me the distance is 1.8 meters. 

Firstly,I chose the on-chip calibration.But,the health check is told me that I don't need to calibrate. I remained to calibrate it by the "RealSense_D400 _Custom_Calib_Paper". In spite of this, there is no changes after the D435 is calibrated. It told me the distance is still 1.8 meters.

Secondly, I chose the tare calibration. And this result of D435 is almost 2.0 meters.However,I find a problem about the calibration. The properties of intrinDepthFrame,intrinColorFrame, extrinDepth2Color seemed unchanged by calibration.

Furthermore,I want to konw how to reset to factory default.Is it hard reset by intel realsense viewer?

MartyG-RealSense, 2020-06-24:
Further information about on-chip and tare calibration can be found in the link below.  

https://github.com/IntelRealSense/librealsense/issues/5838

It describes how Tare "is for correcting the accuracy. You have to point at a flat wall where you know the ground-truth distance to the center point of the image. It will then adjust the calibration parameters accordingly".

You can also calibrate the 400 Series cameras using a tool called the **Dynamic Calibrator**, which can provide an advanced-level calibration.  It will only calibrate the extrinsics, but this should be sufficient to correct mis-calibration problems in most cases.

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

If you are a Linux user, page 14 onward of the user guide for the tool explains how to install in on Linux.

https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide

The Dynamic Calibrator software includes a program called **CustomRW** that can be used to perform a "gold reset" of the camera to its factory configuration.  After launching CustomRW, you can initiate a gold reset by entering the command below.:

**Intel.Realsense.CustomRW.exe -g**

LyJoan, 2020-06-24:
Yes.I point at a flat wall to the center point of the image.The free Dynamic Calibrator is only calibrate the extrin. I don't sure it will help me to calibrate the D435.

MartyG-RealSense, 2020-06-24:
In 99% of cases it is sufficient to just calibrate extrinsics, as these have the most effect on the output image.   There is an OEM version of the software that can calibrate both intrinsics and extrinsics that comes with the $1500 US dollar OEM Calibration Target system sold by Intel.  

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

This is aimed at customers such as engineering departments and factories though, and the free calibrator usually provides a very good re-calibration.

LyJoan, 2020-06-28:
Thank you for your reply and for its speed. The D435 camera got a high depth quality by the  Dynamic Calibrator.
The calibrated parameters after dynamic calibrator is shown here:
[image]
However the result after the tare calibration is shown here:
[image]

As we can see,the intrintDepth and extrinDepth2Color are changed.I still don't konw what calibrated parameters are chaged by Dynamic Calibrator or Tare Calibrator separately.

MartyG-RealSense, 2020-06-28:
Section 2.1 of the user guide for the Dynamic Calibrator software on page 12 explains which extrinsics can be altered by calibration (the free non-OEM version of the software does not calibrate intrinsics):

[image]

My understanding is that Tare is focused on improving the **accuracy** of the camera rather than calibrating it (which is the role of the **On-Chip** self calibration function).

closed 2020-07-18 by MartyG-RealSense