realsenseai/librealsense #6686: How to get Realsense D435 depth field-of-view info?

younussenn, 2020-06-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  {Win (10)  | 
| Platform                                 | PC |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {C# }                          | 
| Segment			|  {others }                          | 

### Issue Description

Hi,
I need depth perspective knowledge with Intel D435 Realsense 2. However, I could not find how to get this information anywhere. I need help with this. How can I get this information with C # code?

MartyG-RealSense, 2020-06-24:
Intel's documentation "cookbook" page of C# example code contains a sample for getting the field of view with C#.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#get-field-of-view

younussenn, 2020-06-24:
Thanks for your answer. I want to learn one more thing. Can we manually adjust the COLOR FOV and DEPTH FOV values? Is there a special method for this?

MartyG-RealSense, 2020-06-24:
The FOVs of the physical imaging sensor components inside the camera are fixed.

If you are seeking for a camera to have the same FOV for depth and color and with the capabilities of the D435, the new D455 depth camera can provide this.  It also has a fast global shutter on the RGB sensor and is matched to the depth FOV so that both have an 86 degree FOV.

https://github.com/IntelRealSense/librealsense/issues/6610

younussenn, 2020-06-24:
Thank you so much!

closed 2020-06-24 by younussenn