realsenseai/librealsense #6688: Noisy point cloud problem with L515

Yibing-Chen, 2020-06-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.00 | 
| Operating System & Version | Ubuntu 18.04 | 
| Platform                                 | PC |
| SDK Version                            | 2.35.2 | 
| Segment			| Sensor | 

### Noisy point cloud problem with L515
  I just compare the point clouds output from L515 and D435i. I found that the point cloud generated by L515 was quite noisy(especially in z direction). Please check the point cloud results below.
  Would someone please tell me what caused the noise with L515? Is there any way to denoise the point cloud? Thanks.

- L515 result:

[image]

- D435i result:
[image]

cdb0y511, 2020-06-24:
It looks like  align problem(depth to rgb).
Can you give a depth frame and rgb frame separately ?

dorodnic, 2020-06-24:
Hi @Yibing-Chen 
The SDK includes several presets for the L515, optimized for different distances and different amount of ambient light (sunlight).
In addition, temporal filter in the SDK can be used to denoise the data. Under camera controls you also have several hardware filtering options. 
Ultimately, the core difference between D400 and L500 is that L500 noise in Z is constant over distance, while with stereo Z noise grows quadratic in distance. Hence, relative noise-to-distance may look worse at really close range, but gets much better as you get further away

Yibing-Chen, 2020-06-25:
Hi @dorodnic , thanks for your reply.
I've just tested the filter options you mentioned by realsense-viewer. It works, but the noise still exists. It seems that the noise is also related to camera orientation. When the camera faces a surface directly, the point cloud is clear, but when the camera orientation changes a bit, the noise appears.

RealSenseSupport, 2020-06-29:
Hi @Yibing-Chen to add to the points already been made, the affects of the surface material can also change how the depth is perceived.  Including the angles in which the camera is pointed at a surface.  If you have the L515 at an angle such that the surface you're pointing to causes the laser to bounce away from the L515, then you'll see differences in the depth data.  You can get better depth data when more of the laser can be reflected back to the L515 itself since time of flight of the laser is key to the technology.

RealSenseSupport, 2020-07-07:
Hi @Yibing-Chen do you need anything further on this?

*If we don’t hear from you in 7 days, this issue will be closed

closed 2020-07-16 by RealSenseSupport