realsenseai/librealsense #6689: A convenient way to install librealsense SDK of a specific release

songshan0321, 2020-06-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | v2.28.0 | 
| Operating System & Version | Ubuntu 16.04 | 
| Kernel Version (Linux Only)    | 4.15.0-106-generic | 
| Platform                                 | PC |
| SDK Version                            | 2.35.2 | 
| Language                            | C++ | 
| Segment			| Robot| 

### Issue Description
I have been using the command below to install librealsense SDK on multiple robots. However, the SDK version is always the latest release, therefore there are different versions of librealsense SDK on my robots.

`apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE || apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u
apt-get install -y librealsense2-dkms
apt-get install -y librealsense2-utils
apt-get install -y librealsense2-dev
apt-get install -y librealsense2-dbg`

Question:
1. Is the librealsense SDK always backward compatible so that I can always do an update on the SDK?
2. If it is not backward compatible, is there a convenient way to install a specific release of SDK so that I can standardize the version on all the robots?

MartyG-RealSense, 2020-06-24:
Hi @songshan0321   If you are accessing an SDK feature that was introduced in a certain version onward, then you would not be able to access that feature in SDK versions older than the one where it was introduced.

If you are using an SDK version that is older than the latest version but is the same or newer than the version where a feature is introduced then there should be back-compatibility.

2.  If you want to ensure that you are installing the same SDK on all robots using Ubuntu then you can obtain a source-code zip file from the SDK **Releases** page, extract its contents and use the extracted folder to build that specific Librealsense version.

https://github.com/IntelRealSense/librealsense/releases/

The source-code zip file can be found in the **Assets** file list at the bottom of the listing of details for each version on the Releases page.

[image]

If you are installing on multiple robots, you may find it convenient to build the source code using the **backend** method.  This requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching.  Once you have extracted a source-code zip file, you can build the SDK from source with the simple method below:

1.  Go to the Librealsense root directory of the extracted folder and create a **Build** folder, and then change to that created folder using the command below (if you have not created the Build folder already):

**mkdir build && cd build**

2.  Whilst in the Build folder, run the CMake command below:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

songshan0321, 2020-06-25:
Hi thank for the reply. May I know what does FORCE_RSUSB_BACKEND do? Is there any convenient way to build on one Linux PC, then I can distribute the build file on other PC with same Linux distribution and kernel. This can avoid building from source on every PC, it takes time.

MartyG-RealSense, 2020-06-25:
**FORCE_RSUSB_BACKEND** installs resources over the internet that would normally have to be patched when building the SDK from source code.  The link below explains the advantages and disadvantages of the backend method compared to patching.  Please scroll down to the section on the link headed '**What are the advantages and disadvantages of using libuvc vs patched kernel modules?**'.

https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604

If you would prefer to be able to deploy easily to multiple devices though, the best way to do so may be to create a GitHub "fork" copy of the Librealsense repository that is not synced with the official repository and so does not update until you deliberately update it.

https://help.github.com/en/github/getting-started-with-github/fork-a-repo

closed 2020-06-26 by songshan0321