realsenseai/librealsense #6690: L515 camera max distance

harmvandeven, 2020-06-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.00 | 
| Operating System & Version |   Win 10 | 
| SDK Version                            |  2.35.2                           | 

When trying to see how the L515 handles the noise on large distances. We just could not get any decent results above the 3 meters. We're just researching if we can use the L515 for a project, so we're not writing any code just yet.

We are simply using the Intel _RealSense Viewer v2.35.2_ and setting the preset to _Max Distance_. But the results we are now getting are far from the advertised 9 meters. Is there a simple toggle we are missing?

Could anybody point us in the right direction?

dorodnic, 2020-06-24:
Hi @harmvandeven 
What type of environment you are working in? (near windows, outdoor, indoor, ...)

harmvandeven, 2020-06-24:
Hi @dorodnic, thanks for your reply.
We are testing inside in our office, there are a couple of windows on the side. We have also tried aiming away from those.

[image]

POWER-KI, 2020-06-24:
I CONFIRM THE PROBLEM

ev-mp, 2020-06-24:
@harmvandeven , can you check that the `threshold filter` is configured beyond 4 meter visibility
[image]

harmvandeven, 2020-06-24:
@ev-mp thanks for your reply. With the L515 camera attached there is no _Threshold Filter_ in the _Post-Processing_ section. When I toggled the _Persistency mode_ to _Always on_ it did (in the end) generate something which reminds me of my studio. But as you can see in the screenshot the _Confidence stream_ is so weak, it takes a while to confidently fill all the pixels. We have played around with the _Confidence Threshold_ but that didn't really help either.

With somebody walking through the studio, I just could not make sense of the generated image.

[image]

RealSenseSupport, 2020-06-25:
@harmvandeven - Have you attempted any other preset?  There is also the No Ambient and Low Ambient presets available.  Can you also provide what the confidence map is showing?  The infrared stream looks saturated as if there is a lot of sunlight flooded into the environment.

harmvandeven, 2020-06-26:
@RealSenseSupport, we have tried a lot of different settings before posting this question, including the _Ambient_ settings. The confident image is shown (the black image, with some specks) on the right of our last picture.
We will try to block the light coming from the windows.

RealSenseSupport, 2020-06-26:
Hi @harmvandeven 

My apologies for not seeing the last image with the confidence map.  Thanks for pointing that out.  Yes I believe there is a lot of ambient light in the room that could be causing issues with determining depth.  Ambient light as well as reflectivity in environments can cause depth confidence to be undesirable.

mjshiggins, 2020-06-27:
Seconding this issue. Range is far below advertised values, even when tweaking all the mentioned presets / settings. "Ambient and reflectivity in environments" -- Does this mean the L515 can only be used in darkness?

blakely3dnm, 2020-06-27:
I have noticed that even indirect, ambient sun light coming in through windows as pictured above strongly impacts the maximum range. Using the sensor at night I was able to register depth measurements much further away.

Another thing to note is that the range specs in the datasheet are @ 95% reflectivity (rather than 80% reflectivity that Ouster reports for their lidar for example) . This means that your surface must have high IR 860 nm reflectivity to achieve that range.  So, that image of the lidar detecting a person 9 meters away on their product page may only be realisitc if that person is wearing high-viz reflectors

AleBasso80, 2020-07-06:
Hello,
I'm experiencing exactly the same issues and I'm very disappointed. Even in a warehouse with a normal/medium environmental lighting the device is not able to detect a white wall at 4-5 m. I hoped that my device was defective, but it's clearly not. In my opinion the L515 can't be used as is in most practical applications.

harmvandeven, 2020-07-13:
@RealSenseCustomerSupport can you confirm the results by @mjshiggins, @blakely3dnm and @AleBasso80. 

> So, that figure on their product page of the lidar detecting a person 9 meters away may only be realistic if that person is wearing high-viz reflectors.

We bought the product for testing based on this video, but (even in a blacked out studio) we've never been able to come near this advertised example. Have you, for clarity, created an example video where you do replicate this advertisement? If not, I would like to ask you to remove the 9 meter claim from the product page, for it is misguiding.

Beamhunter, 2020-07-21:
I'm runing the L515 in anechoic chamber and I have the same problem with the distance, max 3m and the object becomes blurry with no sharp edges.

MichaelTiemann, 2020-07-24:
I will add my voice to the chorus of others who are most interested in long-range use of this device in a studio environment.  I am most interested in what the device can do to measure things in the 2m-9m (or further) range.  Even if 9m is the equivalent of infinity, I would expect 8m to be distinct from that.

RealSenseSupport, 2020-08-04:
We are continuing to improve our documentation and information regarding the L515 camera.  I want to point everyone to a Users Guide that we published that could help in understanding the camera.

https://support.intelrealsense.com/hc/en-us/articles/360051646094-Intel-RealSense-LiDAR-Camera-L515-User-Guide

We are working on more documentation and information regarding the ability to get depth at further ranges within the spec of the camera.

RealSenseSupport, 2020-08-14:
I'd also like to include as far as documentation is concerned the following blog:  https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/

This talks about optimizing for range with the L515 camera

RealSenseSupport, 2020-09-11:
Is there anything else needed on this one?

*If we don’t hear from you in 7 days, this issue will be closed.

whoutman, 2020-09-17:
Yes please! :)

This week we got the L515 camera as well and started testing with the realsense-viewer to see what is possible with the camera. So far, we haven't been able to do proper distance measurements for distances > 3m. I read the information you are referring to in the previous posts, but was wondering if we missed something.

For the experiment I did, I placed some wall-like blocks at approximately 1m increments, obtaining the following measurements:
[image]

In the figure below, the depth information of the same experiment is shown in more detail:
[image]

Within the room, there is no direct sunlight, the preset was set to "Max Range" and the laser power was set to maximum. A low resolution of 640x480 was chosen in order to increase the laser-power per pixel. Increasing the receiver gain did not give significant improvements. The maximum distance threshold filter was set to maximum. What setting could we improve?

Thanks for your help!

MCFurry, 2020-09-22:
Wow! @RealSenseCustomerSupport can you confirm if these issues are solvable by choosing the correct settings?
I really wish to use these L515 sensors for a future projects, but if it can't produce depth images beyond 3 meters in an indoor environment I would be really disappointed... I would definitely expect more from a LiDAR sensor.

AleBasso80, 2020-09-25:
Hello @whoutman and @MCFurry ,
in all my tests I experienced the same issue and eventually I gave up on using the L515 in my projects. Different settings actually change the measurement results, however no configuration seems to be able to deal with an environment that is not dark or very poorly illuminated by the external light, as the sensor saturates in any case. I do hope I'm wrong, but I fear that this issue can't be solved by a new firmware release, but only by some hardware modifications.

RealSenseSupport, 2020-10-02:
Hi @whoutman is there anyway you can provide us with recordings of your session?  Recording with all streams active besides the IMU stream?  

*If we don’t hear from you in 7 days, this issue will be closed.

whoutman, 2020-10-05:
Thanks @AleBasso80 for the info!

@RealSenseSupport We recorded [two files of experiments](https://fschoenm.stackstorage.com/s/zgJmd3kvKOg0MQ86). The readme-file provides the distances to each box. Apart from the settings I changed in my previous [post](https://github.com/IntelRealSense/librealsense/issues/6690#issuecomment-694258844) could you indicate how we can significantly improve the depth information?

Thanks!

RealSenseSupport, 2020-10-05:
Thanks @whoutman for the files.  Let us have a look at it and see if there is more that can be done.

Xenon541, 2020-10-06:
@RealSenseSupport We can confirm that we are likewise not seeing farther than 2.5m is an artificially lit indoor space. No controls or adjustments affect this weakness. Very disappointed as we have a lot of projects we might use this on. Fix it fast!!!

RealSenseSupport, 2020-10-08:
Hi @whoutman 

We've reviewed your bag files, thanks again for that.  What we do see is that the boxes may not be as reflective type material as expected.  Can you let us know if you put high reflectivity target/paper (white paper) on those boxes, how does that change the ability to see the boxes at the various distances.

Also just want to make sure everyone knows that the technology has trade-offs when it comes to environment and object material and reflectivity.  The quality of the return signal impacts the depth data provided by the camera.  Technology overview can be found:  https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/

We are always making updates and one update we did in the latest FW 1.5.1.3 and the latest 2.39.0 LRS is added another depth resolution that might also help your scenario.  We've added QVGA resolution.  The overview blog has been updated with the addition of QVGA.

We are also looking at ways for users to assess their environment and object reflectivity.  Do continue to monitor the latest releases in both FW and librealsense in the future.

@Xenon541 - This information can also be good for your scenario as well.  It is important to note also ambient light can saturate the scenes which can have an impact on depth data and distance.

whoutman, 2020-10-09:
Thanks @RealSenseSupport for the information. We will have a look at it and come back when we updated the firmware and did the tests which are suggested.

whoutman, 2020-10-15:
Hi @RealSenseSupport 

We did some tests after updating the firmware, of which one was with the reduced resolution. You can see them [here](https://1drv.ms/u/s!Ap1AYmQaLilQhOcp3tFc6E3BrBHZDw?e=RNzawB). It seems that the performance of the depth image slightly improved, but given the specs of 9m, we were expecting quite a bit more. 

Here are some pictures of the tests with varying resolution between the two tests. Again, the boxes were placed at 1m increments.
[image]
[image]

RealSenseSupport, 2020-10-23:
Hi @whoutman 

Thanks for the bag files and the information.  We'd like to discuss with you directly about the bag files.  We have some questions.

Can you submit a ticket at the following location?  http://realsensesupport.intel.com/

This will allow us to ask more direct questions with you.

mjshiggins, 2020-10-23:
I’d like to request that this debugging be kept public. This is a problem affecting all users.

whoutman, 2020-11-03:
@RealSenseSupport I just submitted a ticket. 

@mjshiggins: I will put the conclusions here as well as I agree this issue affects all users

whoutman, 2020-11-10:
As promised, I would give feedback on the discussion I had with @RealSenseSupport. 

Based on the measurements we captured, Intel was able to calculate the maximum distance which the sensor is able to measure given the amount of ambient light. To increase the maximum distance with the depth camera, it was advised to reduce the amount of ambient light, reduce the resolution and to select the "max range" preset. 

Our conclusion is that the depth resolution is indeed increased when reducing the amount of ambient light (close to dark!), but this makes the other sensors available on our system useless. In short, for a reasonable amount of ambient light, we don't think the dual functionality (RGB and depth) is reachable at the 9m distance which was specified by Intel.

harmvandeven, 2020-11-12:
@whoutman, thanks for the sharing the feedback. So a small recap for anybody stumbling on this `issue`: The L515 camera could possibly deliver its specs in an ideal setting (at least thats what @RealSenseSupport claims). We've just not seen any actual results in a real world situation.

RealSenseSupport, 2021-01-11:
Wanted to include on this thread some updates that happened regarding range that are important to understanding environment setup:
- https://github.com/IntelRealSense/librealsense/pull/7637
- https://github.com/IntelRealSense/librealsense/pull/7675
These can be found in the latest release of SDK and FW:  https://github.com/IntelRealSense/librealsense/releases/tag/v2.41.0

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-19 by RealSenseSupport