realsenseai/librealsense #6691: best FPS for motionless scene?

EmmanuelCos, 2020-06-24:
| Camera Model                       | D435 | 

Hello,
I am using a D435 cam to capture motionless scenes, and I am wondering what would be the best FPS to use in this case? Are there any other parameters that are recommended to use when capture motionlessscene, to improve depth quality and precision?

Thank you and regards,

Emmanuel

agrunnet, 2020-06-24:
In general the exposure time should be dictated by the lighting. So in normal lighting this may be about 5ms for the d435. That means that you can actually set the frame rate to 90fps and there is no benefit to going to shorter frame rate. However with a motionless scene you should definitely use the post-processing temporal filter and you can decrease the alpha parameter to increase the filtering. This will improve the results about 3x.

MartyG-RealSense, 2020-06-24:
If there is no motion in the scene then 30 FPS should be fine.  If you are using RGB color then there should not be blurring of the RGB image if there is nothing moving in the scene. 

For optimal depth accuracy the **848x480** resolution should be used for D435.

If you find that there is noise or holes in the image then you can improve the quality of the image by applying post-processing filters to perform enhancement functions such as hole-filling.

If possible, avoid fluorescent lights such as ceiling strip lights, which may flicker at frequencies that are hard to see with the human eye and create noise on the image.  The ideal lighting may be LED lights such as those used in photography studios, as doing so should reduce shadows in the scene.

EmmanuelCos, 2020-06-24:
Thank you for your answers! And the exposure time for the depth cameras shouldn't be increased when there is no motion?

EmmanuelCos, 2020-06-24:
I am trying to play with the temporal filter on realsense viewer, modifying the alpha parameter, but don't see any difference at all. Should I reset the cameras or so?

MartyG-RealSense, 2020-06-24:
@agrunnet can give better advice on exposure times than I can.  I would add the following points though:

-  An exposure of 5 ms for normal lighting is suggested above.  In Intel's excellent camera tuning guide for the 400 Series cameras, which @agrunnet co-authored, Section 15 on the bottom of the final page states that in bright sunlight the exposure can be reduced to near **1 ms** (which can also reduce motion artifacts).

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

-  Manually setting exposure may have an effect on FPS.  

  https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

In regard to making changes in the Viewer to the temporal filter, have you enabled the post-processing filters by setting the parent **Post Procesing Filters** option to On (blue color) as well as the temporal filter itself, please?

[image]

MartyG-RealSense, 2020-07-13:
Hi @EmmanuelCos   Do you still require assistance with this case please, or can it be closed?   Thanks!

closed 2020-07-26 by MartyG-RealSense