realsenseai/librealsense #6695: No RealSense devices were found

Chris-Bee, 2020-06-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.926  | 
| Operating System & Version |   Linux Ubuntu 18  | 
| Kernel Version (Linux Only)    | 4.19.81-v7l | 
| Platform                                 | Raspberry Pi4  |
| SDK Version                            | bc64e52c9e128 close to (2.34.0)  | 
| Language                            |  ROS | 
| Segment			|  Robot | 

### Issue Description

The RealSense is not available after the system was started. Running the ROS node results in the warning that no RealSense device was found (see below). Reconnection the RealSense (unplug and reconnect the USB physically) solves the problem in most cases. Since reconnecting physically is not always an option, I'm wondering if there is a method to prevent this issue or an option to restart the driver (trough systemctl, kernel module, etc.). Unfortunately, I could not find a solution to this.

Terminal information:

process[camera/realsense2_camera_manager-1]: started with pid [1081]
[ INFO] [1592200237.375324631]: Initializing nodelet with 4 worker threads.
process[camera/realsense2_camera-2]: started with pid [1082]
[ INFO] [1592200243.640856849]: RealSense ROS v2.2.13
[ INFO] [1592200243.641469571]: Running with LibRealSense v2.34.0
 15/06 05:50:44,697 WARNING [2828411920] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: No data available, err. num: 61
 15/06 05:50:44,697 ERROR [2828411920] (tm-boot.h:39) Error booting T265
[ WARN] [1592200246.711866718]: No RealSense devices were found!
 15/06 05:50:53,743 WARNING [2828411920] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: No data available, err. num: 61
 15/06 05:50:53,743 ERROR [2828411920] (tm-boot.h:39) Error booting T265
[ WARN] [1592200255.759626212]: No RealSense devices were found!

ev-mp, 2020-06-28:
@Chris-Bee, is the camera being recognized by Librealsense SDK tools such as `realsense-viewer` and `rs-enumerate-devices`?
In case it is not - please attach the kernel log as well.

ev-mp, 2020-07-02:
@Chris-Bee , did you manage to collect data for analysis ?
As a preemptive recipe - In case you'll find T265 not being listed with `lsusb` after boot then you can use `uhubctl` utility to power-cycle  USB ports and reestablish connection with device. #4331, #5527, #6596.

Note that this may require RPI firmware upgrade - https://github.com/mvp/uhubctl#raspberry-pi-4b

closed 2020-07-06 by dorodnic

calmelo, 2021-01-19:
Have you solve the problem? How do you solve it?

Chris-Bee, 2021-03-05:
@ev-mp We haven't tried the USB reset you suggested yet. However, @calmelo we started using the camera on an externally powered USB3 Hub and the problem occurred significantly less. Thus I assume it's a power issue that prevents the RealSense to boot completely.

Chris-Bee, 2022-09-27:
@HripsimeS Unfortunately the solution was to switch from an external hub to the main USB ports of the PI and switch the power of the internal hub off and on to get the RealSense to start again. We used `uhubctl` for this and run the power cycling before each start of the RealSense node.

HripsimeS, 2022-09-27:
@Chris-Bee Thank you so much for your reply! I did the instalations of **uhubctl** following this instruction.
https://github.com/mvp/uhubctl

When I run the command **sudo uhubctl** I can see that I have 8 Ports for my camera's USB, you can find in attached

Do I need to **turn off** and then **turn on** all these 8 Ports to be able to run **rs-enumerate-devices** command to see my connected real sense camera? Please let me know how did you managed to make it work eventually :)

[image]

Chris-Bee, 2022-09-27:
@HripsimeS No problem, it can be a bit tricky to find the right parameters, depending on the embedded board, if USB hubs support it, and how hubs might be inter-connected. On the Raspberry PI, we did run this command preventative before each start of our sensor setup.

```bash
# Perform a RealSense Reset
sudo uhubctl --action cycle --location 2
```

I hope this helps. Also, in case this might be of interest to you, our sensor setup and operation were part of a flight stack development which is soon to be open-sourced.
Repo: https://github.com/aau-cns/flight_stack
Publication: https://ieeexplore.ieee.org/document/9849131