realsenseai/librealsense #6701: Please update the "RS Camera" app on Google Play

rafaelspring, 2020-06-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version | Android | 

### Issue Description
The "RS Camera" app is a **very** important debugging tool for us and our customers using Android. It is critical that it continues to be up-to-date on Google Play because asking customers to side-load something custom-built is a show stopper in many cases.

The current version on Google Play is from Nov. 2019 and uses version 2.30.0 of librealsense. That one does not support the L515 for example.

It would be great if this could be updated to the latest version of librealsense.

MartyG-RealSense, 2020-06-25:
Hi @rafaelspring  I have filed a feature request with Intel for your requested upgrade of **RS Camera** on the Google Play Store.

MartyG-RealSense, 2026-10-29:
Case closed due to an LRS version update of the RS Camera app being performed in March 2021 and no further updates to the app made.

closed 2026-10-29 by MartyG-RealSense