realsenseai/librealsense #6702: D435 Pointcloud2 Unit

Joechencc, 2020-06-25:
I have transfer the D435 pointcloud2 to pcl::PointXYZRGB format. Now I have tested the camera and print out the following parameters.

[ INFO] [1593125671.246552758]: X_min:-34.592617
[ INFO] [1593125671.248589572]: X_max:33.729702
[ INFO] [1593125671.249781068]: Y_min:-25.747810
[ INFO] [1593125671.252830113]: Y_max:25.025896
[ INFO] [1593125671.253727208]: Z_min:0.312000
[ INFO] [1593125671.254897008]: Z_max:65.535004

I wonder what is the unit of the point clouds. Like for msg->points[l].x, I guess the unit is cm. Does this mean z_max which is the depth can detect maximally 65 cm range. Or should I tune some parameters?

MartyG-RealSense, 2020-06-26:
Because Z is being reported as 65.535004, this makes me think that these values represent depth in the form of 16-bit depth values rather than conventional distance measurements (which are produced by multiplying the 16-bit depth value by the "Depth Scale" that the camera is currently set to).  If a particular depth scale is used then it may change the measurement units in which the depth is reported (e.g 65.535 instead of 65535).

A RealSense team member explains the concept further in the link below.

https://github.com/IntelRealSense/librealsense/issues/3508#issuecomment-473516941

Joechencc, 2020-06-26:
Yeah, it is understandable now. I wonder the default unit for x and y are still in 0.01 m ?

MartyG-RealSense, 2020-06-26:
I am not certain about the above question.  There is an interesting discussion about RealSense to PCL conversion though in the link below that states XYZ to be in meters.

https://community.intel.com/t5/Items-with-no-label/Converting-RealSense-Pointcloud-to-PCL-Pointcloud/td-p/541957?profile.language=en

MartyG-RealSense, 2020-07-12:
Hi @Joechencc   Do you still require assistance or can tis case be closed, please?  Thanks!

closed 2020-07-18 by MartyG-RealSense