realsenseai/librealsense #6703: apt-repo bionic InRelease' doesn't support architecture 'armhf'

vkuehn, 2020-06-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):
removed as not relevant thought

### Issue Description
following the instructions on [Package Install](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) but it produces this on a Raspberry Pi with Ubuntu Mate

N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' doesn't support architecture 'armhf'

Unfortunatetly compile doesn't work. I will raise another issue for that. Therefore a package would be greatly appreciated

MartyG-RealSense, 2020-06-26:
Hi @vkuehn  As a reply has been provided in https://github.com/IntelRealSense/librealsense/issues/6704 is it okay to close this particular case **apt-repo bionic InRelease' doesn't support architecture 'armhf'** please?  Thanks!

vkuehn, 2020-07-04:
As mentioned there It would be great to get official support packages...however as this is not really an issue it can be closed

closed 2020-07-04 by vkuehn