realsenseai/librealsense #6704: Failed to compile on Ubuntu Mate 18.04 with Raspberry

vkuehn, 2020-06-26:
With the instructions for  [Linux installation](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md##prerequisites) I tried to compile on a Raspberry Pi 3B with Ubuntu Mate 18.04 but failed in some sections

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |  Ubuntu Mate 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-1063-raspi2 #67-Ubuntu SMP PREEMPT Thu May 21 16:09:38 UTC 2020 armv7l    | 
| Platform                                 | Raspberry Pi 3B  |
| SDK Version                            |  2.2.14               | 

gcc version 7.5.0 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04) 
GNU Make 4.1

### Issue Description
The instruction use ./patch-realsense-ubuntu-lts.sh but this failes for me 
./patch-realsense-ubuntu-lts.sh: line 20: ./scripts/patch-utils.sh: No such file or directory

However I have started the compilation with make -j1

It ends in 

./temp/librealsense/src/media/ros/ros_reader.cpp:109:93:   required from here
./librealsense/third-party/realsense-file/boost/boost/function/function_base.hpp:318:13: warning: placement new constructing an object of type ‘boost::detail::function::functor_manager_common<rosbag::TopicQuery>::functor_type {aka rosbag::TopicQuery}’ and size ‘12’ in a region of type ‘char’ and size ‘1’ [-Wplacement-new=]
             new (reinterpret_cast<void*>(&out_buffer.data)) functor_type(*in_functor);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/realsense2.dir/src/media/ros/ros_reader.cpp.o] Error 4
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2

[  1%] Built target fw
[  8%] Built target realsense-file
[  9%] Building CXX object CMakeFiles/realsense2.dir/src/media/ros/ros_reader.cpp.o
CMakeFiles/realsense2.dir/build.make:782: recipe for target 'CMakeFiles/realsense2.dir/src/media/ros/ros_reader.cpp.o' failed
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/realsense2.dir/all' failed
Makefile:129: recipe for target 'all' failed

Is there a way arround it ?

ev-mp, 2020-06-26:
@vkuehn hello,
1- The kernel patches apply to v4l linux driver only. The recommended installation method for RPI is to build with an alternative driver using `cmake .. -DFORCE_RSUSB_BACKEND=true`, so those should not be applied.

2. The specified error
>c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,

is commonly caused by compiler running out of memory, therefore you should verify that you've configured a swap file as noted ([documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md#installation-instructions)).
See a similar report in #2961.

vkuehn, 2020-06-26:
can you kindly provide the correct cmake command ? thought it's not mentioned in the documentation !?

I actually did create the swap and checked it's active. However I will double check once I know corretly how to start the compiler

ev-mp, 2020-06-26:
@vkuehn  hello,
the base cmake command mentioned in the "driverless" installation guides for [Generic LIBUVC](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md) and [RPI_Raspbian](https://github.com/IntelRealSense/librealsense/blob/a3f71797677f7d7bb9c77ad222b20f395028a74e/doc/installation_raspbian.md#install-realsense-sdklibrealsense)  are
`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release` and 
`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true`

*Note that in the latest versions the flag `FORCE_LIBUVC` was deprecated and replaced by `FORCE_RSUSB_BACKEND`.
The original flag is still valid and is used (back-compat) Looking forward we'll gradually update it throughout the documentation.

For including the Python wrapper add `-DBUILD_PYTHON_BINDINGS=bool:true`
The full list of configuration flags is elaborated on [Wiki](https://github.com/IntelRealSense/librealsense/tree/a3f71797677f7d7bb9c77ad222b20f395028a74e/doc#librealsense-wiki) -
https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration

vkuehn, 2020-06-26:
Thanks for that. Unfortunatetly Ubuntu Mate on the Pasberry is not like Raspian and not like Mate on x64. I 'll try out later

there is also [an intel guide](https://dev.intelrealsense.com/docs/using-depth-camera-with-raspberry-pi-3) for realsense on a pi with mate.It referes to the Linux installation.

MartyG-RealSense, 2020-06-26:
Hi @vkuehn   There was a case recently of another RealSense user who tried using MATE with a Pi 3B and experienced memory problems even though they had a swap file set up.

https://github.com/IntelRealSense/librealsense/issues/6530#issuecomment-640150623

In the same discussion, they also tried Ubuntu 18.04 LTS on a Pi 4 and were able to successfully install 18.04 LTS and the Python bindings on it using the **-DFORCE_RSUSB_BACKEND** method and the **-DBUILD_PYTHON_BINDINGS=bool:true** flag suggested by @ev-mp above, though they had problems with additional modules such as NCS2 and OpenVINO.

vkuehn, 2020-06-27:
@ev-mp the compiliation did now work ! It is not clear if I should run the compilation in two step or in just one like 

cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=$(which python)

If python 2.7 >support is needed ?

@MartyG-RealSense 
No this is not related as no one which is not a developer would ever look at compiling from scratch..even those would appriciate maintained package installations

ev-mp, 2020-06-28:
@vkuehn hello, 
>the compiliation did now work ! 

can you provide the part of compilation that didn't work?
>It is not clear if I should run the compilation in two step or in just one like

The above step **configures* SDK build parameters. To **build** (compile) this configuration you use `make` command, i.e:
```sh
$ cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true
$ make -j1
$ sudo make install
```
https://github.com/IntelRealSense/librealsense/blob/a3f71797677f7d7bb9c77ad222b20f395028a74e/doc/installation_raspbian.md#install-realsense-sdklibrealsense

vkuehn, 2020-06-29:
@ev-mp sorry a typo ...the compilation did work now ! all good !

just wanted to now if I have to give all options in one compiliation run or if there should be multiple runs as in the documention looks like that  !?

vkuehn, 2020-06-29:
closing as no point to call this an issue anymore

closed 2020-06-29 by vkuehn

MartyG-RealSense, 2020-06-29:
@vkuehn   Thanks very much for the update.  I'm pleased to hear that you succeeded with compilation.