realsenseai/librealsense #6705: Possible uninitialized local pointer variable "texture" used_ data”

wuhaoqi-hub, 2020-06-26:
When I use save_ to_ ply  creates a class object,it occurs Possible uninitialized local pointer variable "texture" used_ data”,how can i solve this problem?

MartyG-RealSense, 2020-06-26:
If you are using Python, do you have similar errors if you use the **export_ply_example.py** pyrealsense2 sample program that make use of save_to_ply, please?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py

wuhaoqi-hub, 2020-06-27:
I use c++，it has this problem，i don't know how to solve it.

MartyG-RealSense, 2020-06-27:
If you ask this question on your other case about saving to ply then your questions can all be dealt with in one place.  Thank you very much.

https://github.com/IntelRealSense/librealsense/issues/6677

wuhaoqi-hub, 2020-06-27:
ok,finally,i decided to convert the binary ply to ascii ply by programming,I hope that the save_to_ply function can be used by c++ in the future.

closed 2020-06-27 by wuhaoqi-hub

MartyG-RealSense, 2020-06-27:
Thank you very much for the information.