realsenseai/librealsense #6706: Genlock mode 4 is not using Laser ON all the time (and thus the depth quality is poor)

yan796, 2020-06-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Win 10  |
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  SDK 2.0 (build 2.35.2)                         | 
| Language                            |  python                          | 
| Segment			|  others                         | 

### Issue Description
When using the realsense in genlock mode 4, the emitter is ALWAYS alternating ON_OFF whatever the settings. 
For example, the master clock is set to 10fps, the realsense fps to 30fps in genlock mode 4. The realsense is in sync, but the depth is of poor quality. A check of the IR images show there is no dot visible. Increasing the fps of the master clock to 15fps (half the native fps of the realsense) will shows the dots on the ir images, but the camera would not be sync anymore.

Overall, it seems that the emitter is always in ON_OFF mode, so if we are not at the right acquisition frequency, the dot are not visible and the depth is of poor quality. However, the doc here
https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras said the emitter should be always on in this mode which make sense to have the dot visible (and a depth of good quality)

### Request
Can we have the emitter always on in genlock mode ?

for information, trying to force it to be on does not work (below the setting I use)

profile = config.resolve(pipeline)  # does not start streaming
profile.get_device().first_depth_sensor().set_option(rs.option.inter_cam_sync_mode  , 4    )  
profile.get_device().first_depth_sensor().set_option(rs.option.enable_auto_exposure , 1    )  
profile.get_device().first_depth_sensor().set_option(rs.option.emitter_enabled, 1)  
profile.get_device().first_depth_sensor().set_option(rs.option.emitter_on_off  , 0)
profile.get_device().first_depth_sensor().set_option(rs.option.emitter_always_on    , 1)
profile.get_device().first_depth_sensor().set_option(rs.option.laser_power          , 360  )  
profile.get_device().first_depth_sensor().set_option(rs.option.depth_units          , 0.001)

agrunnet, 2020-06-26:
Yes we will fix that. Hopefully the fix will make it into our next release within 3 weeks.

RealSenseSupport, 2020-08-17:
Hi @yan796 Could you please try v5.12.6.0?

RealSenseCustomerSupport, 2020-09-04:
----------------------------------------------

Hi @yan796, Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

yan796, 2020-09-04:
Hi,
really sorry for the delay answering. The new firmware is far better and I think it is almost perfect. However, with the same settings (10fps realsense as slave), I see sometime the emitter being out of sync and thus the depth becoming very sparse. Please see the gif attached, on left is the depth from realsense,  on the right the depth from Colmap (using rgb master camera) for reference, both depth are same scale and same world reference.

gif here:
https://giphy.com/gifs/RindO0nV0zaHcKaklv

Frame 112
[image]

Frame 113
[image]

RealSenseCustomerSupport, 2020-09-22:
----------------------------------------------

Hi,

The issue you're having with synchronization of camera and emitter may be addressed in a new version of FW that is currently being validated. I will look into its release schedule and let you know if it should fix this problem.

John

RealSenseCustomerSupport, 2020-11-11:
----------------------------------------------

Hello,

A brief update on this issue:
According to our engineer who helped to develop the Genlock feature, based on the GIF image, it looks like noise in the trigger signal.

The PWM unit and the image sensor may react differently to noise in trigger signal.
We have seen this issue. Better shielding and proper grounding of the trigger signal should help.

Are you using emitter ON-OFF mode in conjunction with genlock by any chance? It does not look so based on the GIF image.
But if you are, noise in the trigger signal would also negatively affect laser & meta data state synchronization.
You would get frames tagged with wrong laser state in meta-data. Hence the sparse depth.

We have a better version that gracefully handles noise and restores meta-data and laser state synchronization when noise subsides; it is currently in the validation process. When released it should appear in FW v 5.12.10.0 though we do not know the exact schedule for release.

In any case, it does not negate the need to have clean trigger signal as the issue will manifest itself while noise is present.

I hope this helps but let us know your current status with this issue.

John

yan796, 2020-11-19:
Hi John, 
thank you so much for the explanation. It is really nice to have such feedback. I managed with the current behavior of the RealSense in GenLock Mode (v5.12.6.0) to capture 4 cameras in sync, for 100k images. Most of the depth from RealSense are dense so I am happy with the current behavior. This is part of a research paper I submitted recently.

I suspect shielding the wire is what is missing right now. I will work on a better prototype (v2.0) with the shielding done right.

Thanks again,
Yannick

RealSenseSupport, 2020-12-03:
@yan796 Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

closed 2020-12-22 by RealSenseCustomerSupport