realsenseai/librealsense #6710: error: ‘gluPerspective’ was not declared in this scope, when making librealsense V2.35.2

Ddonsy, 2020-06-27:
I have Pi 4, ubuntu 18.04 installed and want to install librealsense V2.35.2 for setting up ROS support (2.2.14). I downloaded librealsense V2.35.2, and ran Cmake, adding some dependencies. It made tee makeflies without any issue. However, when I try to run make it was 82% success and then gave the error "home/ubuntu/librealsense/common/viewer.cpp:1946:9: error: ‘gluPerspective’ was not declared in this scope ... gluPerspective(45, viewer_rect.w / win.framebuf_height(), 0.001f, 100.0f);" and gave me a suggested alternative: ‘glad_glRectiv’.
How can I get  ‘gluPerspective’  working again? Any help is greatly appreciated!
Thanks

MartyG-RealSense, 2020-06-27:
If you are building Librealsense from source, please try using the **backend** installation method.  It requires an internet connection for the installation process but is not dependent on Linux versions of kernel versions and does not need patching.

Once you have extracted the source-code zip file for 2.35.2, you can build the SDK from source with the simple method below:

1.  Go to the **Librealsense** root directory of the extracted folder and create a **Build** folder, and then change to that created folder using the command below:

**mkdir build && cd build**

2.  Whilst in the Build folder, run the CMake command below:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

Ddonsy, 2020-06-27:
Thanks for the reply. Actually, I have resolved it by adding GL headers to the cpp file. However, when I try to open it using real-sense viewer, it works about few seconds fine and the computer restarts. It has happen every time and I don't know what is the reason. Any idea? thanks

MartyG-RealSense, 2020-06-27:
Could you explain more about the "dependencies" that you added, please?

MartyG-RealSense, 2020-06-27:
A cause of crashes or freezes in the Viewer and other programs that rely on OpenGL such as the Depth Quality Tool could be outdated OpenGL graphics drivers.  Updating your computer's graphics drivers may correct such issues.

ev-mp, 2020-06-28:
@Ddonsy hello,
> However, when I try to run make it was 82% success and then gave the error "home/ubuntu/librealsense/common/viewer.cpp:1946:9: error: ‘gluPerspective’ was not declared in this scope ... [...]
How can I get ‘gluPerspective’ working again? Any help is greatly appreciated!

Please make sure that you have the pre-requisite packages installed, OpenGl, specifically 
`sudo apt-get install -y libglu1-mesa-dev` (from [Raspbian installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md#install-packages))
In case you're running Ubuntu 18 and not Raspbian then the following may apply as well - 
`sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev`  (from [Ubuntu Installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites))

ev-mp, 2020-07-02:
@Ddonsy , did you manage to resolve the compilation issue ?

Ddonsy, 2020-07-02:
Yes. I have resolved it by adding GL headers to the cpp file manually. Thanks

closed 2020-07-02 by Ddonsy

milan-r-shah, 2020-08-11:
> If you are building Librealsense from source, please try using the **backend** installation method. It requires an internet connection for the installation process but is not dependent on Linux versions or kernel versions and does not need patching.
> 
> Once you have extracted the source-code zip file, you can build the SDK from source with the simple method below:
> 
> 1. Go to the **Librealsense** root directory of the extracted folder and create a **Build** folder, and then change to that created folder using the command below:
> 
> **mkdir build && cd build**
> 
> 1. Whilst in the Build folder, run the CMake command below:
> 
> **cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

Just to add more info:

The first step in this process is to download the source code of librealsense from the **Releases** page as a zip file. The source code zip file can always be found in the "Assets" list at the bottom of the information listing for each SDK version on the Releases page:

https://github.com/IntelRealSense/librealsense/releases/

[image]

After downloading the zip file, its contents should be extracted so that you have a **librealsense** folder.At this point you can use the **RSUSB** installation method to install librealsense without dependence on Linux versions or kernel versions and without the need for patching. This makes this installation method particularly suited to Arm devices such as Jetson.

Then follow the above-mentioned command (in the quoted reply).

Reference: https://github.com/IntelRealSense/librealsense/issues/6940#issuecomment-665713929