realsenseai/librealsense #6711: Ubuntu 18.04, PI4, T265 realsence viewer crashes and restart the computer

Ddonsy, 2020-06-27:
I have been trying to get this working and after installed and run realsense-viewer, it runs few seconds and then crashes and restart the PI 4. The camera is connected to USB 3 of PI4. Would you be able to tell me how to resolve this??
Thanks

ev-mp, 2020-06-28:
@Ddonsy  hello, 
Providing some kernel logs will be useful to understand the nature of the crash. There are several guides on extracting Linux kernel panic logs, like https://askubuntu.com/questions/104771/where-are-kernel-panic-logs.  
With that being said in several similar cases in the past the root cause was insufficient power provided by RPI to drive the camera rather than the Distro type. To check whether this applies in your case you can connect T265 via a powered USB hub and update whether this would mitigate the crash.

Ddonsy, 2020-06-28:
Thanks ev-mp for the help. Previously, the PI 4 was powered with  an I-Phone USB adapter and it was ok to run the OS but I think not enough to run the camera. I changed it to Samsung USB adapter and it looks like working ok for running both OS and the T265 camera. I think the issue is not enough power to run the Camera. Having a Powered USB hub might solve the issue. I will check. Also, when I use Ubuntu 20.04 with Pi 4 and Ubuntu 16.04 with PI 3, I used the Samsung USB adapter, which was working fine.

closed 2020-06-28 by Ddonsy