realsenseai/librealsense #6712: Dear support, librealsense 2.35.2 with Raspberry PI 4 4GB is not working at boot:

robertocalvi, 2020-06-28:
| | |
|---|---|
|**librealsense**|2.35.2 RELEASE|
|**OS**|Linux|

Please provide a description of the problem

robertocalvi, 2020-06-28:
Dear support, librealsense 2.35.2 with Raspberry PI 4 4GB is not working at boot: 

28/06 10:14:22,108 WARNING [3069497360] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: No data available, err. num: 61  
28/06 10:14:22,108 ERROR [3069497360] (tm-boot.h:39) Error booting T265

If I unplug and replug USB works but only for a couple of minutes.

Cuou you help us?

DreamLFairy, 2020-06-30:
I use 32-bit raspbern system,the same thing happened to me：WARNING [3069497360] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: No data available, err. num: 61.When I switch 64-bit system, the problem is solved.You can try

patrickpoirier51, 2020-07-28:
@DreamLFairy 
Can you specify the 64-bit system, you have used to get the problem is solved?
I am willing to believe it work but I have to prove it myself ;-)

mbobinger, 2020-11-27:
@DreamLFairy @patrickpoirier51 and @robertocalvi
I am having similar issues, have you gained any further experience with 64bit raspian system and the camera, is it working stable now?

closed 2026-10-23 by MartyG-RealSense